prusa3d/PrusaSlicer #2483: Program crashes when exporting config bundle.

WhiteDragonD, 2019-06-08:
### Version
2.0

### Operating system type + version
Windows 7/10

### Behavior
File -> Export -> Export Config Bundle.
PrusaSlicer has stopped working

I was able to Export this config from Slic3r Prusa Edition and import into Prusa Slicer 2.0 but it crashed when trying to Export from 2.0.

I have added output options in Print settings:
{input_filename_base}-{layer_height}mm-{print_time}-{brim_width}Brim_[support_material]Support_[fill_pattern]_{filament_type[0]}_{printer_model}.gcode

and end Gcode Output:
;MarioOverworld
M300 S0 P75
M300 S2637 P150
M300 S2637 P150
M300 S0 P150
M300 S2637 P150
M300 S0 P150
M300 S2093 P150
M300 S2637 P150
M300 S0 P150
M300 S3135 P150
M300 S0 P450
M300 S1567 P150
M300 S0 P450
M300 S2093 P150
M300 S0 P300
M300 S1567 P150
M300 S0 P300
M300 S1318 P150
M300 S0 P300
M300 S880 P150
M300 S0 P150
M300 S987 P150
M300 S0 P150
M300 S932 P150
M300 S880 P150
M300 S0 P150
M300 S1567 P150
M300 S2637 P150
M300 S3135 P150
M300 S1760 P150
M300 S0 P150
M300 S2793 P150
M300 S3135 P150
M300 S0 P150
M300 S2637 P150
M300 S0 P150
M300 S2093 P150
M300 S2349 P150
M300 S987 P150
M300 S0 P600
M300 S3135 P150
M300 S2959 P150
M300 S2793 P150
M300 S2489 P150
M300 S0 P150
M300 S2637 P150
M300 S0 P150
M300 S1661 P150
M300 S880 P150
M300 S2093 P150
M300 S0 P150
M300 S880 P150
M300 S2093 P150
[... 3342 lines cut ...]
retract_length_toolchange = 4
retract_lift = 0.6
retract_lift_above = 0
retract_lift_below = 199
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 35
serial_port = 
serial_speed = 250000
silent_mode = 0
single_extruder_multi_material = 0
start_gcode = M115 U3.4.1 ; tell printer latest fw version\nM83  ; extruder relative mode\nM104 S[first_layer_temperature] ; set extruder temp\nM140 S[first_layer_bed_temperature] ; set bed temp\nM190 S[first_layer_bed_temperature] ; wait for bed temp\nM109 S[first_layer_temperature] ; wait for extruder temp\nG28 W ; home all without mesh bed level\nG80 ; mesh bed leveling\nG1 Y-3.0 F1000.0 ; go outside print area\nG92 E0.0\nG1 X60.0 E9.0  F1000.0 ; intro line\nG1 X100.0 E12.5  F1000.0 ; intro line\nG92 E0.0
toolchange_gcode = 
use_firmware_retraction = 0
use_relative_e_distances = 1
use_volumetric_e = 0
variable_layer_height = 1
wipe = 1
z_offset = 0

[presets]
print = 0.35mm FAST-NAMED
sla_prin

[image]

sgtriggs, 2019-06-21:
What worked for me (long, stupid process);

Load Slicer
Config Wizard
Tick the box for SL1
Finish wizard and close
Re-open Slicer
Select SL1 in the "Printer" dropdown
Quit
Re-open Slicer
Export config bundle (it should work and not crash)
Run Config Wizard again and unselect SL1
Quit and re-open
Export config bundle

Not sure if some of the quit/open steps are required, but it worked for me.

lukasmatena, 2019-06-21:
This is a likely duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2307. If that's the case, the bug should be already fixed in current master branch.

bubnikv, 2019-08-01:
Duplicate of #2307, it will be fixed in the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-01 by bubnikv

WhiteDragonD, 2019-08-01:
I found a quick fix so you can export before updating to the newer version.... it will trick the program into thinking a SLA profile is installed and won't fail on the export.

Open: **C:\Users\USERNAME\AppData\Roaming\PrusaSlicer\PrusaSlicer.ini**
Under **[presets]** where sla_material and sla_print are empty, add the following:

[presets]
filament = Prusa PET
print = 0.35mm FAST-NAMED
printer = Original Prusa i3 MK2.5-MUSIC
**sla_material = Prusa Orange Tough 0.05
sla_print = 0.05 Normal**