prusa3d/PrusaSlicer #2486: Random Diagonal line when exporting to USB

eddie-moser, 2019-06-09:
### Version
_Version of PrusaSlicer used goes here_
2.0.0.0 Win64
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_ Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

Artillery3D Sidewinder X1 Marlin 1.1.9

### Behavior
* _Describe the problem_
When exporting gcode to USB printer will randomly print a diagonal line from outside edge to skirt. This does not happen when printing from SD Card. Issue also does not happen when printing from USB using S3D gcode

* _Steps needed to reproduce the problem_
Export from PS to USB and print
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
Normal Print
* _Actual Results_
Diagonal Line to skirt
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

eddie-moser, 2019-06-09:
[RepBox2_ MegaBracket.zip](https://github.com/prusa3d/PrusaSlicer/files/3269122/RepBox2_.MegaBracket.zip)

eddie-moser, 2019-06-09:
to be clear, its when using a USB Stick

bubnikv, 2019-08-01:
Sorry, but this cannot be PrusaSlicer's fault. PrusaSlicer exports the same file to both USB and SD card, there is no difference on PrusaSlicer's side, but on your machine / firmware side. Closing.

closed 2019-08-01 by bubnikv