prusa3d/PrusaSlicer #2488: Bring back 45 degree rotate buttons please.

x0rtrunks, 2019-06-09:
### Version
PrusaSlicer 2.0

### Operating system type + version
OSX Mojave

### Behavior
The rotation widget is nice and all, but it's far slower to use 90% of the time when I just want to quickly rotate an object by 45 degree increments. I miss the buttons. Please bring them back. Thanks. 

It also appears that the keyboard shortcuts for rotating by 45 degrees have been removed as well.

larsrc, 2020-03-02:
I'd be happy to just have 90 degree rotation. I was hoping double-clicking the rotate icon would do so.

One of the signs of a good workflow program is that common operations are very easy. Rotating by 45 or 90 degrees is much more common than rotating arbitrarily, please make it easy.

neophyl, 2020-03-03:
They are back.  If you use the 2.2.0 rc page up and page down rotate 45 degrees.

bubnikv, 2020-03-03:
> They are back. If you use the 2.2.0 rc page up and page down rotate 45 degrees.

For the clarity. The page up and page down buttons.

bubnikv, 2020-03-20:
Rotation left / right by 45 degrees is now done by page up / down keys in PrusaSlicer 2.2.0.
Closing

closed 2020-03-20 by bubnikv