prusa3d/PrusaSlicer #2493: Crash on Layer Edit with Wipe Tower

WarrenSchultz, 2019-06-10:
### Version
2.0.0+win64-201905201652

### Operating system type + version
Win10 x64

### 3D printer brand / version + firmware version (if known)
Mk3S/MMU2S

### Behavior
Attempting to edit the layer height on an MMU print with wipe block may lead to a crash.

Because the UI does not allow the selection of multiple items and then pushing the Layer Edit button, I tried to do an edit per item. The two objects were both complex multi-part objects, and when I attempted to edit one at a time, it showed this error message, and then crashed.

[image]

Alternatively, the ability to add a modifier mesh for an entire plate, vs. per object would be greatly appreciated (if it isn't already in there and I'm missing it.)

#### Project File (.3MF) where problem occurs

bubnikv, 2019-08-01:
The application did not crash, it informed you that this operation is not allowed.

I am not quite sure what you want to achieve. We do not allow multiple objects with differing layer heights as this would require too many tool changes.

WarrenSchultz, 2019-08-01:
Please re-read the message.

It displayed the error *and then* crashed.

What I am trying to achieve is to adjust the layer height for two of the same multi-material objects at the same place, using the layer editing tool.

Because I can't group-select to edit them all at the same time, I tried to edit one, and then I got the error, and then it crashed after clicking ok.

bubnikv, 2019-08-01:
> What I am trying to achieve is to adjust the layer height for two of the same multi-material objects at the same place, using the layer editing tool.

You should be able to do that if you created two instances of the same object.

The application should not crash.

WarrenSchultz, 2019-08-01:
The problem is that while that would solve that specific case, it wouldn't solve the issue for other uses that I have.
(Think specifically, game tokens which have icons at the same height which I want to have more detail across all pieces.)

lukasmatena, 2019-08-01:
Because the layer editing tool doesn't allow you to edit multiple objects at once, you'll never set precisely the same layer profile twice. The workaround is either print two instances (like @bubnikv suggested) or, if the objects should differ, add one object as a part of the other (which is not nice, but probably the only way). I'm not sure how difficult it would be to allow layer editing for several objects simultaneously, but probably more difficult than one would guess.

As for the crash, that is of course not right, but I cannot reproduce it. Neither in 2.0.0, nor in current master. Could you try opening the console, type "set SLIC3R_LOGLEVEL=9" and then run the application from that console? This will enable extensive logging that might give a clue about the cause of the crash.

WarrenSchultz, 2019-08-02:
I'll give it a shot, thanks.
Honestly, I don't know how the code for parsing the STLs works under the hood, but it would be extremely useful for the future to be able to edit the height globally (plate-wide).

WarrenSchultz, 2019-08-09:
This works perfectly in 2.1.0 alpha 1 with the layer height table. Thank you!

closed 2019-08-10 by WarrenSchultz