prusa3d/PrusaSlicer #2498: Feature Request: Temperature Offest in Printer Settings

Belkaar, 2019-06-11:
I'd like to propose a Temperature Offset Setting in the Printer Settings that gets added to the hotend temperatures when generating g-code.

This would be very usefull when switching between brass and stainless steel nozzle, so you don't need to edit all the filament settings.

I need about 10 more degrees to get the same print result if I'm using a stainless steel nozzle.

beef76, 2021-03-17:
is this feature request in consideration for one of the next updates? was looking for the same and found this request. Since it was opened in 2019 was just wondering if it's on the list for any update soon ;-)

Tupson444, 2026-09-12:
still relevant

Jan-Soustruznik, 2026-10-01:
Hello, thank you for update, According to new [policy](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy) I closing this feature request and converting it into a Discussion thread to hear what the community thinks about this feature. Thank you, and have a nice day.