prusa3d/PrusaSlicer #2499: Feature Request: Combine objects

manfre, 2019-06-11:
Related to #1275 and other oddities caused by Slicer handling each object independently, I'd like the ability to select multiple objects on the build plate and combine them in to a single object. This is a faster version of loading the objects you want combined, exporting build plate as STL, and then re-importing.

rtyr, 2021-02-01:
Implemented in 2.3.0. Closing.

closed 2021-02-01 by rtyr

khenriks, 2022-01-03:
How? I can't find any documentation for this, and it wasn't mentioned in the release notes from 2.3.0.

rtyr, 2022-01-04:
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-alpha1

> One may newly merge multiple objects into one multi-part object by selecting multiple objects in the Plater or in the Object list and by executing "Merge" from the pop-up menu.

hoblin, 2023-09-28:
Is this option available in the Mac version?
[image]

cubuspl42, 2024-09-08:
@hoblin First, separate the instances ("Set as Separated Objects"). Then, select the freshly separated objects, and you'll be able to merge them.