prusa3d/PrusaSlicer #2500: Allow for multiple objects while Spiral Vase mode is on

probonopd, 2019-06-11:
### Version
2.0.0+linux64

### Operating system type + version
xubuntu-18.04-desktop-amd64.iso

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK2S

### Behavior
* Import the attached file to PrusaSlicer
* Switch on Spiral Vase mode
* Click the Add instance (+) icon to get more than one copy onto the build plate
* Error message: "The Spiral Vase option can only be used when printing a single object."

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs

[vase.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3278841/vase.3mf.zip)

#### Intended Behavior

Multiple instances of this object should still be printable, but the distance between the instances needs to be increased so that the hotend does not crash into the other instances.

wavexx, 2019-06-13:
You can actually do that if you enable "complete individual objects" in the print settings -> output options.

probonopd, 2019-06-13:
> You can actually do that if you enable "complete individual objects" in the print settings -> output options.

Wow, awesome, thank you very much. 

I think it would be useful if this information would be contained in the error message. I would never have found this out without your help.

wavexx, 2019-06-13:
It is a good point. We do something similar when switching on supports where, for example, "detect bridging perimeters" is suggested.

However consider that printing multiple complete objects is still a hit and miss. You do have to carefully plan the _order_ in which the objects are printed in order to avoid collisions (the default check is too simple). The bed is not re-heated for the second object, which makes adhesion a bit harder, and some other gotchas...

probonopd, 2019-06-14:
>The bed is not re-heated for the second object

Maybe it would be possible and useful to print the base plate for all objects together like without spiralize, and only thereafter switch from normal mode to spiralize each object individually for the outer walls?

wavexx, 2019-06-14:
If you search for individual/complete mode you'll find several reports about improvements.

I do currently slice object individually and then combine the the gcode by hand when I need to serially print several objects in a row.

bubnikv, 2020-12-09:
> I think it would be useful if this information would be contained in the error message. I would never have found this out without your help.

The error message was updated with  fb7098e542f12864821585b2e248ba1cf5123d26

closed 2020-12-09 by bubnikv