prusa3d/PrusaSlicer #2501: overlapping objects do not get sliced correctly

tcurdt, 2019-06-12:
### Version

Version: 2.0.0+
Build: PrusaSlicer-2.0.0+-201905201652

### Operating system type + version

Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS High Sierra Version 10.13.6 (Build 17G7024)

### 3D printer brand / version + firmware version (if known)

Prusa MK3

### Behavior
* _Describe the problem_

I have added helper discs to a model. They are of course overlapping with the main model. From the preview it seems like it would print overlapping layers.

* _Steps needed to reproduce the problem_

Open the attached project.

* _Expected Results_

The model union should be sliced - not models individually. Spots on the layers should only be printed once.

* _Actual Results_

[image]

_Is this a new feature request?_ No, a bug report.

#### Project File (.3MF) where problem occurs

[NP-FZ100_HolderRemix_001.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3279309/NP-FZ100_HolderRemix_001.3mf.zip)

tcurdt, 2019-06-12:
In comparison S3D does it correctly.

[image]

haarp, 2019-06-12:
As a workaround, you can arrange them how you want, go to File, Export, Export as STL, then add the exported STL to the plater again. The new STL will be treated as a single object.

tcurdt, 2019-06-12:
@haarp as a workaround I sliced it with S3D :) but I hope this will be fixed in the future

bubnikv, 2019-06-14:
If you want to overlap objects, you should load them as parts of an object.
This is not possible in a simple mode, but it is possible in advanced or
expert mode. Just click the right mouse button at the object in the object
list and load another part.

tcurdt, 2019-06-14:
Interesting. Not quite so straightforward but glad it's possible.

bubnikv, 2019-08-01:
The difference between S3D and PrusaSlicer is, that S3D slices the whole plater at once, while PrusaSlicer slices each object separately. This allows PrusaSlicer for example to chose different layer height for each object.

Could you please close this issue?

wavexx, 2019-08-01:
I've been passively watching this. I don't think that PS does anything wrong here.

If I can put some thought, maybe what's needed is some feedback in the normal view. Similarly to how we change the color to blue when an object doesn't fit, changing the color of the two overlapping objects to something else (red?) would be useful.

This could also pave the way of some form of warning in the future ("overlapping objects detected: convert them to parts?").

tcurdt, 2019-08-01:
@bubnikv That is an explanation of the cause - but frankly speaking not a good reason to close the issue (IMO). The least it should do is detect the overlap and give a warning. Right now it is slicing it in a non-printable manner - which I would argue is the wrong behaviour.

tcurdt, 2020-07-11:
"Add part" works but when I click on "add instance" it duplicates the whole object - not just the selected part.
A work around is to use c&p of the part.

JamesKR3D, 2020-07-25:
I have to agree with tcurdt.. I don't see any reason overlapping parts should be sliced separately causing overlapping extrusions which can lead to clogs.  If there is a reason for it, please enlighten us.  It would be better to give a notification when it happens and slice all of it as a single object.

MadMax389, 2022-12-09:
S3D is not doing it correctly either. The disc should not be part of the model, but like a brim with its own outline so it is easy to remove. Watch this video and it explains how to do it in Prusa Slicer. https://www.youtube.com/watch?v=-FlPlq7zkqM 
[image]

tcurdt, 2022-12-09:
"The disc should not be part of the model, but like a brim with its own outline so it is easy to remove"

I guess that's debatable. When I only add these disc I want really good adhesion. Using the brim style provides much less of that. These disc are really is easy to cut off. So I think I personally prefer them as part of the model.
But I can see the argument for the brim style.

MadMax389, 2022-12-10:
You can still easily do it that way in Prusa Slicer. Just add the disc as a part from the gallery (or user .stl), and don't change the perimeter value of the disc.
[image]

closed 2025-03-06 by Jan-Soustruznik