prusa3d/PrusaSlicer #2502: Not generating raft beneath entire object

haarp, 2019-06-12:
Version 2.0.0

PrusaSlicer is sometimes not generating sensible rafts. It tends to leave areas unsupported, or in extreme cases, half the object unsupported. See screenshots, AMF attached, please rename from .zip to .zip.amf

[image]
[image]

[idler_mount_v2.zip](https://github.com/prusa3d/PrusaSlicer/files/3280259/idler_mount_v2.zip)

neophyl, 2019-06-13:
I downloaded your amf and placed it into my install of slicer 2.0.0 with a raft of 1 layer and it puts a raft under both parts correctly.  It doesnt look like the one in your screencap though as its more rectangular without the bits that protrude on the sides.

On yours that isnt working save it as a project (3mf file) and post that.  That will store all your model data along with all your settings, model placement etc which would probably help in tracking down the issue as then we should have the exact same settings.

haarp, 2019-06-13:
@neophyl 

Weird, the amf does contain all the settings just like a 3mf would. When I load it into PrusaSlicer 2.0.0, it loads all the settings I had when I first encountered the problem, and upon slicing, can reproduce the problem.

neophyl, 2019-06-13:
afaik the amf doesnt save any of the project or slicer settings with SlicerPE.  the 3mf definitely does though.  Also I think there's a few issues in for strange amf behavior anyway lol.

with a default 0.15 Quality MK3 profile modified to have a single layer raft with 0.175mm elephants foot compensation it does change the raft shape slightly but it looks fine for me.

[image]

haarp, 2019-06-13:
@neophyl Ok, PrusaSlicer is weird. Sometimes it imports the config from the amf, sometimes it doesn't. The config is definitely in the file tho, as you can extract the zip, then open the .amf in a text editor to see it.

I am attaching the config manually here. Import using the menu. 
[config.ini.zip](https://github.com/prusa3d/PrusaSlicer/files/3285094/config.ini.zip)

neophyl, 2019-06-13:
Ok with that ini file I now get the same issue.  I made the problem go away by changing 1 setting though.  You have Support>Contact Distance set to 0.  Is this intentional as that will make supports VERY hard to remove.  In fact that has a comment on it (Soluble) as its really only used for soluble support material in multi material setups usually.
If you change it to the more normal 0.2 (Detachable) option and re-slice it then behaves more normally.  0.1 works too but I have found that supports are still difficult to remove at 0.1.

haarp, 2019-06-13:
>  Is this intentional as that will make supports VERY hard to remove.

Yes, this is intentional. I'm printing these parts out of Nylon, which has bed adhesion problems. I'm using a PETG raft to overcome this problem. PETG and Nylon separate from each other nicely.

neophyl, 2019-06-13:
ah ok.  Not had the need of anything beyond PETG as yet.  In which case I'm not sure if I can suggest anything else to help.
The weird behaviour only seems to happen with the contact distance of 0, setting 0.001 and it slices fine.  It might be worth trying the 0.001 value and printing to see if that will work for you adhesion wise.

bubnikv, 2019-06-14:
You seem to have two issues:

1) You exported an amf from PrusaSlicer 2.0.0 and the AMF does not contain project configuration. Also the exported AMF is not zipped, while it should be.

Going forward, we will only support 3MF as a project file, though there seem to be regression issues.

2) Missing raft below the 2nd object.

This is weird, I have no explanation. Anyway, fixing the 2nd object with the netfabb STL fixing service or cutting just a tiny bit from the bottom of the object fixes the problem. I am not sure whether it is a model issue or slicer issue.

haarp, 2019-06-14:
@bubnikv I am very confused now.

> You exported an amf from PrusaSlicer 2.0.0 and the AMF does not contain project configuration.

It does! If you open the .amf in a text editor, you can see the project config under `<metadata type="slic3rpe_config">`. But PrusaSlicer doesn't seem to load it when loading the AMF.

> Also the exported AMF is not zipped, while it should be.

Are you referring to `idler_mount_v2.zip` attached above? You have to rename it to `idler_mount_v2.zip.amf` first. I had to call it `.zip` because Github wouldn't allow `.zip.amf` attachments.

> Going forward, we will only support 3MF as a project file, though there seem to be regression issues.

How do I export that? PrusaSlicer 2.0.0 only offers export as amf or stl, not 3mf. I compiled it myself, did I maybe miss a configure option?

> Missing raft below the 2nd object.
> This is weird, I have no explanation. Anyway, fixing the 2nd object with the netfabb STL fixing service or cutting just a tiny bit from the bottom of the object fixes the problem. I am not sure whether it is a model issue or slicer issue.

I'm pretty sure it's nto a model issue. I've seen this problem with other models aswell.

Setting elephant foot compensation to a different value (e.g. 0 or 0.2mm) fixes it. Rotating the object 90° gives a different, but also broken raft.

Whatever is broken seems to depend on the area or shape of the bottom of an object.

bubnikv, 2019-06-14:
AMF files should be compressed by a zip algorithm by default from slicer
already.

> How do I export that? PrusaSlicer 2.0.0 only offers export as amf or stl,
not 3mf. I compiled it myself, did I maybe miss a configure option?

Just use file -> Save Project.

haarp, 2019-06-14:
> AMF files should be compressed by a zip algorithm by default from slicer already.

Correct. And that was exactly what I attached ;) I had just renamed it so Github wouldn't complain.

> Just use file -> Save Project.

Ah, easy enough. I only considered the Export menu. Here's the 3mf attached. This time I *additionally* zipped it so Github wouldn't complain.

[idler_mount_v2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3291421/idler_mount_v2.3mf.zip)

bubnikv, 2019-11-01:
Elephant foot compensation shall not be applied onto the object's first layer if it is printed over the raft. That is certainly an error in the first place.

bubnikv, 2020-01-02:
The elephant foot compensation improvements are part of PrusaSlicer 2.2.0-alpha2. Closing.

closed 2020-01-02 by bubnikv