prusa3d/PrusaSlicer #2503: Start End Overlap Perimeters

touradnik9, 2019-06-12:
### Version
_Version of PrusaSlicer used goes here_
Slic3r Prusa Edition
Version: 1.41.3+win64
Build:   Slic3rPE-1.41.3+win64-full-201902111723

_Use `About->About PrusaSlicer` for release versions_
Start End

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
Can a feature be added for a start and stop overlap for perimeters? Thanks!!!

_Is this a new feature request?_
YES
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2019-06-14:
>  Can a feature be added for a start and stop overlap for perimeters?

Would you expand on that?

touradnik9, 2019-06-15:
Sure,

For example, I am trying to fix this gap, as seen in the screenshot in Slic3r.  Some other slicers, like Mattercontrol, have a feature called 'Start End overlap' for perimeters.  I think this feature is great! Can it be added?

Thanks!

[image]
[image]

cringegnere, 2023-04-22:
+1 for this feature, It would be nice to add it in the next 2.6.0 alpha release!

georgiys, 2023-06-09:
+1 for this feature! 
It become critical with big nozzles. It's very hard to make a seam with 0.6 nozzle. It forms just a break. Especially If you have a problem with a filament, if it's less than have to be: less material extruded, perimeter end can't reach a start.

AlekseyMko, 2024-06-28:
+1
would be very helpful!

lukasmatena, 2024-11-21:
Added in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena