prusa3d/PrusaSlicer #2504: Implement Drop to Plate for Modifiers

ctilley83, 2019-06-12:
### Version
2.0.0

### Operating system type + version
Mac OS X 10.14.5

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
There is no way to drop a modifier to the plate.  Currently the behavior is a default modifier box is placed on the bed.  Resizing it causes the modifier to float in midair.  To move a modifier in the Z axis, you have to use sliders or input the Z coordinates.  This creates two problems.

1. The origin is set to the middle of the parent part. So setting Z=0 places the modifier mid air.
2.  The other option is to use the sliders, however you can't precisely place it on the plate forcing you to eyeball it.

The use case this would solve is if you want to apply different settings to PRECISELY the first 5mm of the part. The simple solution is implement the "Place on Face" feature for modifiers. It seems easy enough to implement.

_Is this a new feature request?_
I guess it's a feature request but I'm sitting here in awe that this was overlooked.

ctilley83, 2019-06-16:
Would anyone like to discuss?

bubnikv, 2019-06-17:
@lukasmatena has it on his plate. It is a low hanging fruit to harvest.

ctilley83, 2019-06-17:
@bubnikv Awe man, that's a shame.  It's the easiest way to apply different settings at different layer heights with PrusaSlicer.  There needs to be a better way unless I'm missing something obvious.

lukasmatena, 2019-06-17:
@ctilley83 If you're asking about yet easier way to use different print parameters for selected layer spans, that will be implemented in future (as mentioned in e.g. https://github.com/prusa3d/PrusaSlicer/issues/2102).

As a quick solution for now, we are going to implement a simple drop to plate button to make life easier - which is what you originally suggested, so I'm not sure I catch your drift.

ctilley83, 2019-06-17:
Adding a simple drop to plate makes my workaround easier.  It works in my particular use case because I only need to make one setting change during the print.  Issue #2102 will fix it permanently. I didn't know that issue existed so I apologize for that.  I still see the ability to drop modifiers to the plate helpful in other use cases though.

bubnikv, 2019-08-01:
This feature will be implemented in the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-01 by bubnikv

ctilley83, 2019-09-04:
This never happened and we are on RC.  Additionally, referenced issues as well as announcements by Josef mentioned implementing ways to add different print settings to the print based off of certain layers.  For example S3D processes where you can change infill settings, number of perimeters etc without using modifiers.

lukasmatena, 2019-09-05:
I'll try to be as explicit as you apparently need me to:

From PrusaSlicer2.1.0-alpha1 release log:
> The modifier meshes could newly be dropped to the print bed using a new button at the "Part manipulation" side panel #2504.

Let us try it:
[image]

As for the layer height modifier, let me quote from the very same change log of PrusaSlicer2.1.0-alpha1 release log:
>  One can now not only define a layer height for a Z span, but one can also apply any parameter to the Z span of an object, that can be assigned to object's modifier mesh. 

And indeed:
[image]

I suggest reading the release logs. @bubnikv spends a lot of time writing them, there is no reason anyone else should spend time searching in them for you.

ctilley83, 2019-09-11:
@lukasmatena  Ok I missed the drop to plate feature of the modifiers but the button is so small I needed a magnifying glass to see it. It would be easier if you could use the drop to plate feature in the main view like you would a normal model. One thing I noticed, is you're not consistent in the coordinate system for modifiers.  You have a Y is up configuration which is fine because that's how it's done in the cinematic world, but 3D printing uses Z is up.

As far as the layer editing release log item, I can't make out what the living hell you're talking about.  Maybe it's a translation issue since English is not your first language, but nobody in their right mind can interpret that.

bubnikv, 2019-09-12:
> : As far as the layer editing release log item, I can't make out what the living hell you're talking about. Maybe it's a translation issue since English is not your first language, but nobody in their right mind can interpret that

Would you please give an example?
English is certainly not my mother's tongue, but I hoped it is not that bad.