prusa3d/PrusaSlicer #2505: Splitting an STL causes parts to fall down and there's no way to move em across Z axis

Goury, 2019-06-12:
### Version
2.0.0+linux64

### Behavior
Load an STL, right click green thing, click split → to objects.
Expected Results: everything stays where it was, just being split into multiple objects.
Actual Results: everything falls down to the bottom.
Bonus Results: there's no way to change Z position, it just doesn't let you.

Surely splitting to parts doesn't have such problem, but then I can't edit those parts, so it makes this feature useless for this case.

Proposal: let us change Z position and let us not drop everything down.

WilkoV, 2019-06-12:
What do you try to do?

Goury, 2019-06-12:
To split a multipart STL and then to cut some parts of it

neophyl, 2019-06-13:
when you choose to split it to parts you are telling SlicerPE that they are separate objects so it naturally places them all on the bed to make them printable.

Model editing should be done in model editing software not a slicer.

Goury, 2019-06-13:
Why are there model editing tools then?
What if I don't want to print something on the bed?

Don't make this harder for everyone.
It is clear that placing everything on the bed should be optional, especially when it was not on the bed in the first place.

neophyl, 2019-06-13:
I think you may be missing an option in SlicerPE.  There are 2 split options, split to Objects and split to parts.  When you choose to Split to Object you are explicitly telling SlicerPE that it should treat them as separate objects.  So it does treat them as if you had imported them in as separate stl files.  

So they get placed on the bed as you cant print an object that's in mid air.  After all you can't print something that isnt connected to the bed in some fashion.
This behaviour is perfectly logical as that's the option you have choosen.

Splitting to parts leaves them where they are and they are still part of the object.  This can be seen in the tree hierarchy on the right.

It's great that you want to use Slicer for everything but sometimes its better to use the right tool for the right job.

Goury, 2019-06-13:
One way to resolve this issue would be to enable editing of parts.

But placing everything on the bed is still not right and should be optional because I can and do want to print some stuff not on bed.
Software shouldn't dictate us how to use our machines.

bubnikv, 2019-06-14:
> Surely splitting to parts doesn't have such problem, but then I can't
edit those parts, so it makes this feature useless for this case.

What do you mean with editing those parts?

> Proposal: let us change Z position and let us not drop everything down.

Why? The problem we are facing is that we don't understand what you are
trying to achieve with the software.

Goury, 2019-06-15:
> What do you mean with editing those parts?

Cutting and rotating them

> Why?

I wan't my machine to print something not on bed.
Why not?

Surely under no circumstances this shouldn be default, nor should it be not in the advanced section.
But why are we so against printing some parts not from Z0?

neophyl, 2019-06-15:
Sorry but how can a 3D printer print something that is not located on the bed ?

Goury, 2019-06-15:
It can print it not on the bed.
For example, I can put something on the bed.
For another example, it can print a lot of spagetti into a box.

Also anything is possible with imagination.

But the real question is why are you so against imagination?

neophyl, 2019-06-15:
So I ask an hones6 question about how it’s physically possible to print something in the air and you turn it into a personal attack on my lack of imagination ?  Great. 

For the record I have plenty of imagination it’s just that if I wanted to print something with one object on top of another I would open up my modelling software of choice, Blender in my case, model the awesomely fantastic things I have imagined and then I would import it in the SLICER and let it do its job of slicing the part to create the G-code. Where in that is a lack of imagination ?  

I don’t expect my modelling software to be able to create gcode the same as I don’t expect my slicer of choice to be able to model. 

Also with split to parts you can move parts around and modify them include place them in the air in the z axis so to me it looks like the whole ‘issue’ is a lack of understanding on your part on how to actually use the very powerful software tools that we have been given.  For free. 

I come here to check issues and to possibly help other users because that’s just good behaviour but when it descends into character assassination then it becomes not to be fun anymore. I wish you well in your future printing endeavours. Maybe you can find the perfect slicer for you, or maybe you will end up writing one. Anything is possible with enough imagination as you point out.  Good luck.

Goury, 2019-06-15:
> and you turn it into a personal attack

No, I didn't.
But you did right now by falsely accusing me in personal attack.

> I have plenty of imagination it’s just that if I wanted to print something with one object on top of another

Not enough imagination yet.
Imagine now that you don't want another object to be printed.

> Also with split to parts you can move parts around and modify them include place them in the air in the z axis

How do I do that?
I set Z to something else and it does nothing.
Also, if that is possible, why does it reset Z position on splitting then?

The very point of that topic is that this doesn't work.

neophyl, 2019-06-15:
Add object model that has several separate parts,  choose split to parts NOT split to objects. You can then move the separate parts around by selecting then from the tree on the right. You have to select them from the tree though and not by selecting them from the build plate window.  You still have to have one of the parts placed on the bed. I don’t care how much imagination you apply you can not print an object in mid air.  You will get spaghetti I think you called it. If I wanted to print a tangled mass of filament I wouldn’t need a slicer, I could just write some gcode to move the extruded in mid air while extruding in any text editor. 

Also accusing someone of a lack of something when you know nothing about them IS a personal attack.

bubnikv, 2019-08-01:
@Goury 

> But the real question is why are you so against imagination?

We are trying to make an easy to use software. Your proposal may make sense to you, but to the majority it will make the software less intuitive and it will just risk creating spagetti monsters.

@neophyl gave a description of how you may move parts of an object in Z.

Closing.

closed 2019-08-01 by bubnikv

zalexzperez, 2020-03-31:
I was trying to print a design in 2 colors that would be similar to a Prusa lcd cover with engraved text whose bottom will be in another color. 

Since the front part will be touching the bed and I DON'T want the whole layer to be in a different color, but only the bridge infill in this case, I have to divide my design in 3 parts.

I first tried Split to objets and found this topic. Now I tried Split to parts but I can't hide the different parts. You can only hide different objects. Why is this?

m-sundman, 2023-02-15:
@Goury is absolutely right, and @neophyl is completely, utterly wrong, to say it as nicely as possible. There are plenty of situations where one could want to split an object into 2, and keep the parts where they are. If you automatically shuffle parts around then at the very least make that optional. I didn't ask the slicer to move the things around so it shouldn't. If I want to move some object then I can move that object.

And @neophyl said that it should be possible to move objects off the bed and @bubnikv  apparently agrees.
**_How???_**
Whenever I try to change Z nothing happens, exactly as @Goury says. There's a "drop to bed" button next to the Position fields, but objects already automatically drop and stay glued to the bed surface and there seems to be no apparent way to avoid this, so why does this button even exist? It is extremely frustrating when the software is actively working against you in futile attempts at "being smart" by doing things I neither asked it to do nor want it to do.

lukasmatena, 2023-02-16:
@m-sundman Please, read #1513 and https://help.prusa3d.com/en/article/how-to-lift-object-from-the-print-bed_245192.

Goury, 2023-02-16:
@lukasmatena this only confirms that prusa added yet another useless anti-feature because of delusional fantasies.

Closing a bug just because it's hard to fix is just an insult.

lukasmatena, 2023-02-16:
> Closing a bug just because it's hard to fix is just an insult.

It is not my fault that you are filing duplicates. #1513 is open, nobody closed it because it is "hard". And even then it would not be an insult.

I can recommend reading https://help.prusa3d.com/en/article/how-to-lift-object-from-the-print-bed_245192 or https://github.com/prusa3d/PrusaSlicer/issues/1513#issuecomment-952048965. If you actually care.

m-sundman, 2023-02-19:
> @m-sundman Please, read #1513 and https://help.prusa3d.com/en/article/how-to-lift-object-from-the-print-bed_245192.

The first is really long, mostly filled with people saying that it's really, really dumb that objects can't be lifted up. I agree with those people. The second link shows how to lift sub-objects, which might work in some situations, but in many situations will not work. The justifications on that page are invalid and...how do I say this nicely... not very well thought out.

There are so many situations where you might want to slice something not touching the build plate. There's no way any programmer would be so dumb that they couldn't think of any, and if there is then they should be fired for just basic incompetence.

E.g., let's say my previous print failed 22h in. I want to continue it. I wish to slice just the top part again, at the correct height. The top part to be printed will NOT be touching the build plate, but rather be printed on top of the part that was already printed. Just let me cut the model at the appropriate height, and keep the top part where it is, "floating in the air", and slice it, and let me print it right there "in the air". Your "make a multi-part object in which only the bottom part touches the build plate" won't work, because the bottom part can't be printed at all.