prusa3d/PrusaSlicer #2506: Crash while manually editing SLA supports.

BeldrothTheGold, 2019-06-12:
### Version
_Version of PrusaSlicer used goes here_
2.0.0-win64 and tested with latest git commit at time of report. 

 

 ~/git/PrusaSlicer (master) $ git log                                                                                                                       
  commit abdb5c5d742f17457f5fe0bec572f1fa668e6344 (HEAD -> master, origin/master,  origin/HEAD)        

                     

### Operating system type + version
PrusaSlicer
Version:   2.0.0+win64
Build:     PrusaSlicer-2.0.0+win64-201905201652

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18362), 64-bit edition

<details>
  <summary>OpenGL installation</summary>

GL version:   4.6.0 NVIDIA 430.39
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 2070/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
[... 439 lines cut ...]
 PrusaSlicer.dll!Slic3r::CLI::run(int argc, char * * argv) Line 484
 	at c:\users\scott\git\prusaslicer\src\prusaslicer.cpp(484)
 PrusaSlicer.dll!slic3r_main(int argc, wchar_t * * argv) Line 663
 	at c:\users\scott\git\prusaslicer\src\prusaslicer.cpp(663)
 prusa-slicer-console.exe!wmain(int argc, wchar_t * * argv) Line 296
 	at c:\users\scott\git\prusaslicer\src\prusaslicer_app_msvc.cpp(296)
 [External Code]
</details>

Example scene immediately prior to crash occurring: 
[image]

Where the assertion fails while debugging: 
[image]

 Values of int i and m_editing_mode_cache when the crash occurred while debugging from:
 PrusaSlicer.dll!Slic3r::GUI::GLGizmoSlaSupports::select_point(int i) Line 1102
 	at c:\users\scott\git\prusaslicer\src\slic3r\gui\gizmos\glgizmoslasupports.cpp(1102)
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Can happen with any object while editing the SLA supports.

BeldrothTheGold, 2019-06-13:
This definitely looks like a race condition with m_hover_id.

Just got a similar crash in another method:

PrusaSlicer.dll!Slic3r::GUI::GLGizmoSlaSupports::on_update(const Slic3r::GUI::GLGizmoBase::UpdateData & data, const Slic3r::GUI::Selection & selection) Line 714
	at c:\users\scott\git\prusaslicer\src\slic3r\gui\gizmos\glgizmoslasupports.cpp(714)

<details>
  <summary>Stack Trace</summary>
PrusaSlicer.dll!Slic3r::GUI::GLGizmoSlaSupports::on_update(const Slic3r::GUI::GLGizmoBase::UpdateData & data, const Slic3r::GUI::Selection & selection) Line 714
	at c:\users\scott\git\prusaslicer\src\slic3r\gui\gizmos\glgizmoslasupports.cpp(714)
PrusaSlicer.dll!Slic3r::GUI::GLGizmosManager::update(const Slic3r::Linef3 & mouse_ray, const Slic3r::GUI::Selection & selection, const Slic3r::Point * mouse_pos) Line 259
	at c:\users\scott\git\prusaslicer\src\slic3r\gui\gizmos\glgizmosmanager.cpp(259)
PrusaSlicer.dll!Slic3r::GUI::GLGizmosManager::on_mouse(wxMouseEvent & evt, Slic3r::GUI::GLCanvas3D & canvas) Line 641
	at c:\users\scott\git\prusaslicer\src\slic3r\gui\gizmos\glgizmosmanager.cpp(641)
PrusaSlicer.dll!Slic3r::GUI::GLCanvas3D::on_mouse(wxMouseEvent & evt) Line 2649
	at c:\users\scott\git\prusaslicer\src\slic3r\gui\glcanvas3d.cpp(2649)
PrusaSlicer.dll!wxAppConsoleBase::CallEventHandler(wxEvtHandler * handler, wxEventFunctor & functor, wxEvent & event) Line 672
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\appbase.cpp(672)
PrusaSlicer.dll!wxEvtHandler::ProcessEventIfMatchesId(const wxEventTableEntryBase & entry, wxEvtHandler * handler, wxEvent & event) Line 1398
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\event.cpp(1398)
PrusaSlicer.dll!wxEvtHandler::SearchDynamicEventTable(wxEvent & event) Line 1868
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\event.cpp(1868)
PrusaSlicer.dll!wxEvtHandler::TryHereOnly(wxEvent & event) Line 1589
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\event.cpp(1589)
[Inline Frame] PrusaSlicer.dll!wxEvtHandler::TryBeforeAndHere(wxEvent &) Line 3873
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\include\wx\event.h(3873)
[Inline Frame] PrusaSlicer.dll!wxEvtHandler::ProcessEventLocally(wxEvent &) Line 1526
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\event.cpp(1526)
PrusaSlicer.dll!wxEvtHandler::ProcessEvent(wxEvent & event) Line 1499
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\event.cpp(1499)
PrusaSlicer.dll!wxEvtHandler::SafelyProcessEvent(wxEvent & event) Line 1617
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\event.cpp(1617)
PrusaSlicer.dll!wxWindow::HandleMouseEvent(unsigned int msg, int x, int y, unsigned int flags) Line 5550
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\window.cpp(5550)
PrusaSlicer.dll!wxWindow::HandleMouseMove(int x, int y, unsigned int flags) Line 5633
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\window.cpp(5633)
PrusaSlicer.dll!wxWindow::MSWHandleMessage(__int64 * result, unsigned int message, unsigned __int64 wParam, __int64 lParam) Line 3014
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\window.cpp(3014)
PrusaSlicer.dll!wxWindow::MSWWindowProc(unsigned int message, unsigned __int64 wParam, __int64 lParam) Line 3758
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\window.cpp(3758)
PrusaSlicer.dll!wxWndProc(HWND__ * hWnd, unsigned int message, unsigned __int64 wParam, __int64 lParam) Line 2847
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\window.cpp(2847)
[External Code]
[Inline Frame] PrusaSlicer.dll!wxWindow::MSWSafeIsDialogMessage(tagMSG *) Line 2711
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\window.cpp(2711)
PrusaSlicer.dll!wxWindow::MSWProcessMessage(tagMSG * pMsg) Line 2595
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\window.cpp(2595)
PrusaSlicer.dll!wxGUIEventLoop::PreProcessMessage(tagMSG * msg) Line 145
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\evtloop.cpp(145)
PrusaSlicer.dll!wxGUIEventLoop::ProcessMessage(tagMSG * msg) Line 163
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\evtloop.cpp(163)
PrusaSlicer.dll!wxGUIEventLoop::Dispatch() Line 230
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\msw\evtloop.cpp(230)
PrusaSlicer.dll!wxEventLoopManual::ProcessEvents() Line 228
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\evtloopcmn.cpp(228)
PrusaSlicer.dll!wxEventLoopManual::DoRun() Line 268
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\evtloopcmn.cpp(268)
PrusaSlicer.dll!wxEventLoopBase::Run() Line 76
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\evtloopcmn.cpp(76)
PrusaSlicer.dll!wxAppConsoleBase::MainLoop() Line 380
	at c:\users\scott\git\slic3r\deps\build\dep_wxwidgets-prefix\src\dep_wxwidgets\src\common\appbase.cpp(380)
[... cut ...]
PrusaSlicer.dll!Slic3r::CLI::run(int argc, char * * argv) Line 484
	at c:\users\scott\git\prusaslicer\src\prusaslicer.cpp(484)
PrusaSlicer.dll!slic3r_main(int argc, wchar_t * * argv) Line 663
	at c:\users\scott\git\prusaslicer\src\prusaslicer.cpp(663)
prusa-slicer-console.exe!wmain(int argc, wchar_t * * argv) Line 296
	at c:\users\scott\git\prusaslicer\src\prusaslicer_app_msvc.cpp(296)

</details>

Scene at time of crash:
[image]

Debug window with watches at time of crash:
m_hover_id = 7603082
m_editing_mode_cache {size =43}
[image]

BeldrothTheGold, 2019-06-13:
After some additional debugging, I'm pretty sure the issue is in: 
 PrusaSlicer.dll!Slic3r::GUI::GLCanvas3D::_picking_pass() Line 3806
	at c:\users\scott\git\prusaslicer\src\slic3r\gui\glcanvas3d.cpp(3806)

I set a conditional break point in:
PrusaSlicer.dll!Slic3r::GUI::GLGizmoBase::set_hover_id(int id) Line 161
	at c:\users\scott\git\prusaslicer\src\slic3r\gui\gizmos\glgizmobase.cpp(161)

To break if id is != -1.

Occasionally _picking_pass is calling set_hover_id() with an out of bounds value. I don't know enough about open GL to determine what might be causing the issue. Sorry!

Here's an example of one of the strange values. 
[image]

lukasmatena, 2019-06-13:
Thanks for your investigation, I'll look into it. All I can say at this point is that I wasn't able to reproduce the crash. I also tried to add a range check on the 1102 line and it never threw on my Linux machine.

BeldrothTheGold, 2019-06-14:
After playing with this some more, I realized that the only time I was seeing an out of bounds ID was when hovering over the very edge of a gizmo. This made me suspect multi-sampling was causing the issue. 

It seems my OpenGL drivers don't completely disable my GPUs multi-sampling when glDisable(GL_MULTISAMPLE) is called.  Disabling multi-sampling using NVIDIA Control Panel for prusa-slicer.exe seems to have resolved the out of bounds issue I was seeing. 

Here is the conditional break-point I was setting while debugging this:
Set this break point and then roll your mouse over the very edge of a gizmo somewhere along a diagonal line and when the program breaks, watch the value of id. You should be able to catch the issue. (Assuming your system faces it)
[image]

I found this forum post which sheds some light on a possible fix?

[howto-disable-multisampling-for-stencil-buffer](https://community.khronos.org/t/howto-disable-multisampling-for-stencil-buffer/58593)

bubnikv, 2019-06-14:
Thanks for heads up. This is very bad.

https://www.khronos.org/opengl/wiki/Framebuffer_Object
these are only supported since OpenGL 3.0, but I suppose your graphics driver supports that, right?

Ideally we would do the picking on the CPU using 3D search structures like octree or kdtree and ray casting, but it would require quite a lot of refactoring.

bubnikv, 2019-06-14:
We may try a little trick: We may set the last component of the color to let's say a random multiple of 16. Then if the picked color was interpolated, then the last component will likely not be a multiple of 16. Also we shall test for the out of bounds explicitly if we know that the multi-sampling cannot be disabled reliably. Also on graphics cards that support frame buffers, we shall do the picking on them.

BeldrothTheGold, 2019-06-15:
It looks like there were extensions to OpenGL since 1.1 to support FBOs but those have since been deprecated in favor of the core implementation in 3.0.

https://www.khronos.org/registry/OpenGL/extensions/EXT/EXT_framebuffer_object.txt

Also, most if not all desktop graphics processors manufactured after 2011 have suppoted OpenGL >3.0 including Intel's integrated graphics processors. AMD and Nvidia graphics cards manufactured after about 2009 have supported OpenGL >3.0.

https://en.wikipedia.org/wiki/List_of_Intel_graphics_processing_unitss
https://en.wikipedia.org/wiki/List_of_AMD_graphics_processing_units
https://en.m.wikipedia.org/wiki/List_of_Nvidia_graphics_processing_units

BeldrothTheGold, 2019-06-15:
So it looks like there have been extensions to OpenGL since 1.1 to provide framebuffer objects. 

https://www.khronos.org/registry/OpenGL/extensions/EXT/EXT_framebuffer_object.txt
https://www.khronos.org/opengl/wiki/Framebuffer_Object_Extension_Examples

Though, as the link you provided mentions, the extension was been deprecated in OpenGL 3.0 and dropped from the API since 3.1 in favor of the core implementation. 

Also, most desktop graphics processors have supported OpenGL since 2011. Including Intel's integrated GPUs. 

https://en.wikipedia.org/wiki/List_of_Intel_graphics_processing_units
https://en.wikipedia.org/wiki/List_of_AMD_graphics_processing_units
https://en.m.wikipedia.org/wiki/List_of_Nvidia_graphics_processing_units

BeldrothTheGold, 2019-06-16:
Check this out, I added a debug option to render the picking pass to the screen. 

https://github.com/prusa3d/PrusaSlicer/compare/master...BeldrothTheGold:master

If you look closely, the purple and blacks are being interpolated.

[image]

BeldrothTheGold, 2019-06-16:
Here is the same with "Antialiasing - Mode" set to "Off" in the NVIDIA Control panel:

[image]

enricoturri1966, 2019-06-17:
Working on it -> https://dev.prusa3d.com/browse/SPE-964

BeldrothTheGold, 2019-06-18:
Works for me. You guys rock!

closed 2019-06-18 by BeldrothTheGold

bubnikv, 2019-08-09:
We were able to reproduce the picking issue on @YuSanka 's computer. Interestingly, this issue was masked by another issue, where the PrusaSlicer 2.0 had a flag disfunctional, which worked for Slic3r PE before, instructing the dual graphics card setup to switch to the high power graphics card (NVidia in her case). Once she switched to NVidia, the picking issue surfaced, which was very disturbing in rectangle selection (hold shift and drag mouse with left mouse button down), as many objects out of the selection were added into the selection.

It took couple of iterations, but I was able to add a random enough checksum into the alpha channel, by which I am able to reject anti-aliased or multi-sampled pixels with a very high probability, making the rectangular selection working again on @YuSanka's computer.

a58b1844e032fc58e13a115a3fabedfd0690a04f
2ad3c05a655cd801dc88adf3af7a4429a47ee6b3
2d7f478dac6206284da6aed9c75a111a0a1db2b7
a19a506ad8d925f9c9a6df93311f8b87b7205012
7cef1292b20306eb19cede4fefce492c590dfedf
841d0796b79b923f98cd477b39c2abf9ab6e7ac4

BeldrothTheGold, 2019-08-09:
@bubnikv, @YuSanka 
I appreciate the update. I remember you discussing doing something like this.
Hopefully this is the last we see of this issue.