prusa3d/PrusaSlicer #2508: No feature layers showing in slice view

Geoffree3, 2019-06-13:
### Version
2.0.0+-201905201652

### Operating system type + version
MacOS 10.14.5

### 3D printer brand / version + firmware version (if known)
Wanhao i3plus

### Behavior
After slicing (or selecting slice view), the platter is empty.  The layer navigator (vertical scroller) is not visible.  Only when 'travel' is selected is anything shown and the vertical scroller, but this only shows travel between printing!

#### Project File (.3MF) where problem occurs
[VR headset side 6s.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3285390/VR.headset.side.6s.3mf.zip)

neophyl, 2019-06-13:
On your Printer Settings tab you have G-Code flavour set to no extrusion.

That's why your model uses 0.00 filament and takes 39 minutes to 'print' and only travel moves are shown.  You've basically selected travel moves only.

[image]

Geoffree3, 2019-06-13:
Hi

Many thanks for pointing that out.  I was pulling my hair out looking!   I must have fiddled with that when I was setting the slicer up initially and then of course the setting didn’t show up when I was in advanced user mode.   I did wonder why it said 0.00 filament  :-)

Thanks again
Geoff

bubnikv, 2019-06-14:
would you please close the issue if it has been resolved for you?

bubnikv, 2019-08-01:
Wrong settings was used, the issue seems to be resloved. Closing.

closed 2019-08-01 by bubnikv