prusa3d/PrusaSlicer #2513: 3MF Import Bug

marcus2tts, 2019-06-13:
### Version
Version 2.0.0+win64
Windows10 Pro
Version 1703 OS Build 15063.1418

Prusa MK3s with MMU2s

When I import 3MF and AMF files all the separate bodies are forced down onto the plate and are not treated as a group.  The older versions of slcr PE did not do this. 

The only workaround I see is to go to one of the bodies in the tree and right click onto it's settings icon and re-add the original amf file to it under "Add part". Then you have to delete all the original bodies. I consider this a big interface problem. 

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[Concentric Rings.zip](https://github.com/prusa3d/PrusaSlicer/files/3287835/Concentric.Rings.zip)

bubnikv, 2019-06-14:
It looks like the same issue as in
https://github.com/prusa3d/PrusaSlicer/issues/2317 , right?

marcus2tts, 2019-06-14:
Yes, that is the same issue and same workaround.

bubnikv, 2019-08-01:
This issue will be fixed by the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

rtyr, 2020-11-03:
Fixed. Closing.

closed 2020-11-03 by rtyr

OddlyCreative, 2023-01-09:
Hi, still having this issue using PrusaSlicer version 2.5.0 on Windows 64. Attached with the google drive link is the .3mf I am attempting to import; viewed on other software it appears as normal. 

[image]

[image]

https://drive.google.com/drive/folders/1pOJlUxE4fLIluKY3wcsrKN4D8_7bKWLa?usp=sharing