prusa3d/PrusaSlicer #2514: Add Brim and/or setting to increse footprint of Brim for MMU2

m00dawg, 2019-06-14:
### Version
2.0.0

### Operating system type + version
Linux Mint 18.3

### 3D printer brand / version + firmware version (if known)
Prusa MK2s, MMU2s, Latest Firmware

### Behavior
* For tall prints (e.g. 185mm), the wipe tower seems too narrow and can be knocked over by the nozzle, particularly after reheating the nozzle after waiting for MMU2 attention (e.g. to have it reload a filament - it seems to retry too few times in my case though that's another issue). The can cause the surface of the tower to end up being rough and due to the speeds of some of the wiping operations, it's a lot of force being applied to the tower and is too fast to merely remelt and resurface the plastic.
* While Prusa Slicer has options to change the width, this is in relation to the volume. Meaning I can only make a narrow tower, or a square one, but can't actually intentionally print a bigger tower.
* Likewise I could not find ANY options to change the brim width of the tower itself, which would otherwise considerably improve bed adhesion.
* There also are no speeds I can find to slow down the wipe tower operations to help smooth out bumps caused by restarting, etc.
* For short models, this isn't as much of an issue, but for tall ones, the nozzle doing its then acts more and more as a lever on the tower such that, the taller it is, the more force is being applied to the bottom increasing the changes it will pop off the bed. Unfortunately this only happens for me 75% of the way into a tall print.
* The only workaround I can find is to try increasing adhesion. I will try Magigoo in hopes it helps. I am printing in PLA (Atomic) and, of note, I have no bed adhesion issues with normal prints. Just the wipe tower.

_Is this a new feature request?_
Yes and no. I would call it a half bug since there is no software workaround otherwise.

#### Project File (.3MF) where problem occurs
https://www.prusaprinters.org/prints/1765-dragon-wine-glass-multi-material

jasontibbitts, 2019-06-20:
As a corollary, for short prints the current purge block can seem to be too "heavyweight"; the bottom really doesn't need to be solid.  I do a large number of prints of objects which are only a few mm thick, often with only a single filament change.  The tower doesn't really need a solid bottom in that case.

So I guess one solution is to provide a choice of purge block types, or to otherwise provide some additional control over how it is generated.

iamacup, 2020-03-22:
Hey, so this is the top result for ‘add brim to purge block’ and a variety of other google results – here is a solution for people that are experiencing issues with the block coming off on long prints or whatever.
Just grab a cube STL, and manipulate the dimensions so that it is 1 layer (or two, if you want) high and shove it over the space the purge block is on the bed like this:

[image]

And you will get a slice that looks like this:

[image]

With a first layer like this:

[image]

rtyr, 2021-02-02:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/4149.

pacaj2am, 2021-02-13:
not working with latest prusaslicer. The object is generated over the wipe tover... see:

[image]

lukasmatena, 2021-03-05:
Wipe tower brim width will be adjustable in PrusaSlicer 2.4.0 (1efa9a06f67d7796bda21fedcff7bdabadbd66a5).
Closing.

closed 2021-03-05 by lukasmatena

erickwill, 2021-03-05:
@Lukáš Matěna

*Any forecast for PrusaSlicer 2.4.0 or a beta?*

*Thanks*

lukasmatena, 2021-03-06:
@erickwill No forecast. It will be released when it's ready.