prusa3d/PrusaSlicer #2515: Segfault on startup when run under Wayland

jasontibbitts, 2019-06-14:
### Version
2.0.0

### Operating system type + version
Linux, Fedora 29, using the distribution-supplied packages.  Compiled with wx 3.0, using GTK3.  (Fedora does not have GTK2 versions of wx 3, only wx 2.8).

### Behavior
I don't use Wayland myself, but a user reported that when run under Wayland, prusa-slicer segfaults immediately on startup.  I think this, like #1963, is probably related to building with the GTK3 version of wxWidgets.  Since it does appear that you're tracking GTK3 issues, I figure I would go ahead and file this to let you know.
```
(gdb) where
#0  0x00007ffff2b49455 in xcb_take_socket () at /lib64/libxcb.so.1
#1  0x00007ffff5a806ee in  () at /lib64/libX11.so.6
#2  0x00007ffff5a80f3d in _XFlush () at /lib64/libX11.so.6
#3  0x00007ffff5a83afd in _XGetRequest () at /lib64/libX11.so.6
#4  0x00007ffff5a77051 in XQueryExtension () at /lib64/libX11.so.6
#5  0x00007ffff5138d82 in  () at /lib64/libGLX.so.0
#6  0x00007ffff5134da9 in glXQueryVersion () at /lib64/libGLX.so.0
#7  0x00007ffff7211ee5 in wxGLCanvasX11::GetGLXVersion() () at /lib64/libwx_gtk3u_gl-3.0.so.0
#8  0x00007ffff7212f15 in wxGLCanvasX11::ConvertWXAttrsToGL(int const*, int*, unsigned long) ()
    at /lib64/libwx_gtk3u_gl-3.0.so.0
#9  0x00007ffff721364c in wxGLCanvasX11::InitXVisualInfo(int const*, __GLXFBConfigRec***, XVisualInfo**) ()
    at /lib64/libwx_gtk3u_gl-3.0.so.0
#10 0x00007ffff721377a in wxGLCanvasBase::IsDisplaySupported(int const*) () at /lib64/libwx_gtk3u_gl-3.0.so.0
#11 0x0000555555c90384 in Slic3r::GUI::GLCanvas3DManager::_detect_multisample(int*) (attribList=0x7fffffff91f0)
    at /usr/src/debug/prusa-slicer-2.0.0-2.fc29.x86_64/src/slic3r/GUI/GLCanvas3DManager.cpp:265
#12 0x0000555555c905dc in Slic3r::GUI::GLCanvas3DManager::create_wxglcanvas(wxWindow*) (parent=parent@entry=0x555556973d30)
    at /usr/src/debug/prusa-slicer-2.0.0-2.fc29.x86_64/src/slic3r/GUI/GLCanvas3DManager.cpp:227
#13 0x0000555555d05eb8 in Slic3r::GUI::View3D::init(wxWindow*, Slic3r::GUI::Bed3D&, Slic3r::GUI::Camera&, Slic3r::GUI::GLToolbar&, Slic3r::Model*, Slic3r::DynamicPrintConfig*, Slic3r::BackgroundSlicingProcess*)
    (this=0x555556973d30, parent=<optimized out>, 
    bed=..., camera=..., view_toolbar=..., model=0x5555566fa198, config=0x5555567046c0, process=0x5555566fb068)
    at /usr/src/debug/prusa-slicer-2.0.0-2.fc29.x86_64/src/slic3r/GUI/GUI_Preview.cpp:52
#14 0x0000555555d061c6 in Slic3r::GUI::View3D::View3D(wxWindow*, Slic3r::GUI::Bed3D&, Slic3r::GUI::Camera&, Slic3r::GUI::GLToolbar&, Slic3r::Model*, Slic3r::DynamicPrintConfig*, Slic3r::BackgroundSlicingProcess*) (this=0x555556973d30, parent=0x5555566c2d30, bed=..., camera=..., view_toolbar=..., model=0x5555566fa198, config=0x5555567046c0, process=0x5555566fb068) at /usr/src/debug/prusa-slicer-2.0.0-2.fc29.x86_64/src/slic3r/GUI/GUI_Preview.cpp:34
#15 0x0000555555b11712 in Slic3r::GUI::Plater::priv::priv(Slic3r::GUI::Plater*, Slic3r::GUI::MainFrame*) () at /usr/src/debug/prusa-slicer-2.0.0-2.fc29.x86_64/src/slic3r/GUI/Plater.cpp:1437
#16 0x0000555555b1281b in Slic3r::GUI::Plater::Plater(wxWindow*, Slic3r::GUI::MainFrame*) () at /usr/include/c++/8/ext/new_allocator.h:86
#17 0x0000555555ae7198 in Slic3r::GUI::MainFrame::init_tabpanel() () at /usr/src/debug/prusa-slicer-2.0.0-2.fc29.x86_64/src/slic3r/GUI/MainFrame.cpp:180
#18 0x0000555555af0802 in Slic3r::GUI::MainFrame::MainFrame() () at /usr/src/debug/prusa-slicer-2.0.0-2.fc29.x86_64/src/slic3r/GUI/MainFrame.cpp:69
#19 0x0000555555adb464 in Slic3r::GUI::GUI_App::on_init_inner() (this=0x5555561b4540) at /usr/src/debug/prusa-slicer-2.0.0-2.fc29.x86_64/src/slic3r/GUI/GUI_App.cpp:238
#20 0x0000555555adb8db in Slic3r::GUI::GUI_App::OnInit() (this=<optimized out>) at /usr/src/debug/prusa-slicer-2.0.0-2.fc29.x86_64/src/slic3r/GUI/GUI_App.cpp:157
#21 0x00007ffff7c48d32 in wxEntry(int&, wchar_t**) () at /lib64/libwx_baseu-3.0.so.0
#22 0x00005555558343fd in Slic3r::CLI::run(int, char**) () at /usr/src/debug/prusa-slicer-2.0.0-2.fc29.x86_64/src/PrusaSlicer.cpp:484
#23 0x00005555558253a1 in main () at /usr/include/c++/8/ext/new_allocator.h:79
```
In the older Fedora slic3r-prusa3 packages, we patches the slic3r.pl script to add `$ENV{GDK_BACKEND} = 'x11';`.  For the Fedora prusa-slicer package, I'm going to use a shell wrapper that does the same.

lukasmatena, 2020-09-01:
Just for reference, the solution proposed above  was implemented and PrusaSlicer Fedora package does not currently crash on Wayland. Same path was taken today by OpenSUSE maintainers (#4691).

bubnikv, 2020-10-09:
We are now setting GDK_BACKEND environment to "x11" at the start of PrusaSlicer on Linux, which should fix this issue.
The change will be in PrusaSlicer 2.3.0-alpha1.
Closing.

closed 2020-10-09 by bubnikv