prusa3d/PrusaSlicer #2517: Feature Request - posibility move modificators across models, or swap/change stl

fundix, 2019-06-15:
### Version
2.0.0+

Hi, I am printing huge statue with many of similar parts. I have minimal infil and I need add some in fill modificators, but if I want print new similar part I have to create new modificators from the beginning.

It's possible move modificators across models?
Or change whole stl file and keep modificators?

[image]

bubnikv, 2019-06-17:
You can copy / paste modifiers across models, just use the Ctrl-C / Ctrl-V
hot keys.

fundix, 2019-06-17:
no, realy? :-D Thank you :-D

closed 2019-06-17 by fundix

adamdport, 2021-01-23:
@bubnikv this doesn't appear to work for me. While I _can_ copy/paste modifiers across models, it only works if it's a different instance the exact same model. If the model is changed at all, the modifiers don't position correctly. 

Can @fundix reopen this, or should I create a new feature request? Any ideas how to resolve this? Thanks.

adamdport, 2021-01-23:
@bubnikv this solution is also not helpful now that we have other modifiers like seam painting that can't be copied/pasted afaik. I think it would be ideal to have the "reload from disk" functionality, only from _another model_. This way, when I make a minor change to the model, I don't need to start from scratch in my prusaslicer project. I'll start a new feature request.