prusa3d/PrusaSlicer #2518: Crashes when giving part empty name

haarp, 2019-06-15:
2.0.0

Repro:
- Have part
- Click on part name once, rename mode becomes active and selects the entire string
- Press del
- Click anywhere to apply the name change

-> PrusaSlicer will crash

Thank you!

PS, I'd also like to say that triggering a rename operation when single-clicking on a part name once is very bad usability. In 99.9% of all times I have another operation but renaming in mind when doing so.

BeldrothTheGold, 2019-06-15:
Hello Haarp,

What operating system and version are you using?

`### Operating system type + version`

Also, I am not able to reproduce this crash on the latest commit on my system (Windows 10). 

Also, on my system clicking on the name once only selects an item. I have to click again on the selected item in order to activate remaining.

BeldrothTheGold, 2019-06-15:
Here is a quick GIF from my PC. 

[image]

haarp, 2019-06-15:
Hello,

oh sorry, I should've added that. I'm on Gentoo Linux under X11. Gentoo is a rolling release, so there are no versions. I compiled PS myself.

And you're right, it only starts renaming on click when the item already was selected. But if it is, a single click will trigger it.

bubnikv, 2019-06-17:
What wxWidgets and what GTK did you compile against?

haarp, 2019-06-17:
> What wxWidgets and what GTK did you compile against?

gtk+-3.24.8
wxGTK-3.0.4

bubnikv, 2019-06-17:
Would you try to compile against gtk2?

haarp, 2019-06-17:
> Would you try to compile against gtk2?

Yes, I just did by configuring with -DSLIC3R_GTK=2. PS looks different, but still crashes with the same backtrace.

pmjdebruijn, 2021-03-01:
I've just been hit by this on git master as well:

```
ASSERT INFO:
slic3r-prusa-2.3.0~377+gb6cd41d67/src/slic3r/GUI/ExtraRenderers.cpp(32): assert "variant.GetType() == "DataViewBitmapText"" failed in operator<<().

BACKTRACE:
[1] wxDataViewRendererBase::FinishEditing()
[2] wxDataViewRenderer::FinishEditing()
[3] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[4] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[5] wxEvtHandler::TryHereOnly(wxEvent&)
[6] wxEvtHandler::ProcessEventLocally(wxEvent&)
[7] wxEvtHandler::ProcessEvent(wxEvent&)
[8] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[9] wxTextCtrl::GTKProcessEvent(wxEvent&) const
[10] wxWindow::GTKHandleFocusOutNoDeferring()
[11] wxWindow::GTKHandleFocusOut()
[12] g_closure_invoke
[13] g_signal_emit_valist
[14] g_signal_emit
[15] gtk_widget_send_focus_change
[16] g_cclosure_marshal_VOID__OBJECTv
[17] g_signal_emit_valist
[18] g_signal_emit
[19] g_signal_emit_valist
[20] g_signal_emit
[21] gtk_widget_grab_focus
[22] g_closure_invoke
[23] g_signal_emit_valist
[24] g_signal_emit
[25] gtk_cell_area_stop_editing
[26] g_signal_emit_valist
[27] g_signal_emit
[28] g_cclosure_marshal_VOID__BOXEDv
[29] g_signal_emit_valist
[30] g_signal_emit
[31] gtk_event_controller_handle_event
[32] g_closure_invoke
[33] g_signal_emit_valist
[34] g_signal_emit
[35] gtk_main_do_event
[36] g_main_context_dispatch
[37] g_main_loop_run
[38] gtk_main
[39] wxGUIEventLoop::DoRun()
[40] wxEventLoopBase::Run()
[41] wxAppConsoleBase::MainLoop()
[42] wxEntry(int&, wchar_t**)
[43] __libc_start_main
```

It's trivially reproducable on any wxGTK build that doesn't have asserts manually forced off.

YuSanka, 2021-03-03:
> 
> It's trivially reproducable on any wxGTK build that doesn't have asserts manually forced off.

I have a build with a wxWidgets assertes allowed, but can't reproduce this issue

pmjdebruijn, 2021-03-03:
Hey, I just rebuilt using wxGTK 3.1.4-patched, and the issue is indeed no longer reproducible.

So it seems this is a wxGTK 3.0.x specific issue.

closed 2026-09-29 by prusaslicer-bot[bot]