prusa3d/PrusaSlicer #2519: Takes focus away from input box when moving cursor to plater

haarp, 2019-06-15:
2.0.0

When I attempt to enter something into an input box (e.g. Part Manipulation) and move the mouse cursor onto the plater, keyboard focus is instantly taken away from the input box. This very frequently leads to keyboard input accidently triggering random functions in the plater, which can result in loss of work. It's also bad usability.

Thank you!

haarp, 2019-08-11:
In 2.1.0 Alpha-1, Backspace now deletes objects. Together with this bug, this continuously leads me to accidently delete parts I want to edit. Luckily there's Undo now, but it's still super annoying.

lukasmatena, 2023-02-01:
Fixed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena