prusa3d/PrusaSlicer #2521: Error exporting G-code on Linux to some file systems

ardenpm, 2019-06-16:
### Version
PrusaSlicer Version 2.0.0+linux64

### Operating system type + version
Ubuntu 16.04 x86-64

### Behavior
Exporting G-code to a file system which does not support chmod operations causes an error message _Copying of the temporary G-code to the output G-code failed_ to be displayed. The G-code file is written if it is the first time the file is being written to, however if subsequently exporting to the same file it is not correctly overwritten.

To reproduce.

1. Must be running on a Linux system and must have a file system mounted that does not support chmod operations. The simplest is a Samba share.

2. Load a model (any model), slice it and press the _Export G-code_ button.

3. Select the mounted file system as the destination and provide a filename then save.

4. Observe the above mentioned error message. Note that the file was exported correctly.

5. Repeat the process but change a setting to produce new G-code and save to the same file again.

6. Note that the old file is not replaced with the new G-code.

The expected result is that the error is not displayed and the file is always overwritten if requested. The actual result is that an error message is displayed and exported files are not overwritten/updated.

Note that on native Linux file systems this issue does not occur. Also note, here is some example strace output showing the offending system call which seems to be causing the problem.

```
[pid  9759] chmod("/run/user/1000/gvfs/smb-share:server=qnap,share=3dp/LAYZOR/tmp.gcode", 0644) = -1 EOPNOTSUPP (Operation not supported)
```
On Ubuntu it seems related to how users are handled on Samba shares. They are not mapped to local usernames correctly so the chmod operations fail, even though the user can read, write and delete the file.

Perhaps not being able to chmod should be ignored if the file can still be written correctly.

bubnikv, 2019-06-17:
>  Perhaps not being able to chmod should be ignored if the file can still
be written correctly.

I think that is what we need to do.

gavanfantom, 2019-09-03:
I'm also seeing this when saving gcode directly to a FlashAir card over DAV. Same OS and version of PrusaSlicer.

Jebtrix, 2019-09-03:
There is a parameter **noperm** that silently ignores these commands from linux on a CIFS mount. Thats easy enough to define in fstab but with gnome gfvs I'm not sure. 

Just some Ubuntu context: [how-can-i-tell-gvfs-to-use-noperm-chmod-problem-on-gvfs-smb-share](https://askubuntu.com/questions/1110550/how-can-i-tell-gvfs-to-use-noperm-chmod-problem-on-gvfs-smb-share)

lf-, 2019-10-25:
Well I hit and fixed this bug. Hope the pull request goes well ^_^

lf-, 2019-10-25:
So I fixed this bug and reintroduced the failure to fix permissions. Ooooops. Debugging before I send in the pull request.

commit 964d7bb8 (2019-11-01): Fix g-code export to permissionless filesystems

Fix #2521
Fix #3102
commit d729315d (2019-11-01): Fix g-code export to permissionless filesystems

Fix #2521
Fix #3102
closed 2019-11-01 by vojtechkral