prusa3d/PrusaSlicer #2522: Prusaslicer hangs up if the program is not on my main monitor.

GeusB, 2019-06-16:
### Version 2.0.0+win64

### Windows 10 Home 1803 Build 17134.829

### Prusa I3 Mk3: 3.7.1-2266

### Prusaslicer hangs up if the program is not on my main monitor.
When I try to slice any STL (tried about 4 straight from thingiverse), with or without changing print or other settings, it always hangs. CPU/RAM are at about idle levels. If I slice the same files with PS on my main monitor they work perfectly. 

#### [Kitchen_Utensil_Holder_Final.zip](https://github.com/prusa3d/PrusaSlicer/files/3294203/Kitchen_Utensil_Holder_Final.zip)

Extra Info
PrusaSlicer
Version:   2.0.0+win64
Build:     PrusaSlicer-2.0.0+win64-201905201652

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17134), 64-bit edition

OpenGL installation
GL version:   4.6.0 NVIDIA 417.35
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 970/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
[... 307 lines cut ...]
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

bubnikv, 2019-06-17:
Weird. It is very difficult to debug such an issue without access to such a
setup.

Is your computer a laptop with dual graphics? (Intel for low performance
and nVidia for high performance tasks)?

GeusB, 2019-06-17:
It's a desktop pc with a separate graphics card. (but is does also have an onboard gpu)
 
It's not that it's a big issue(to me at least), I just moved PS back to the main monitor.
But if you require extra/more detailed info I'll gladly help.

Braddg, 2019-06-18:
I have the same issue, but I can't even get PrusaSlicer to open on non-main monitor.  It hangs trying to display the 3D build sheet.  I do not have built-in graphics.  I have a desktop with two NVIDIA 1080Ti graphics cards, driving three monitors.

I am able to use Slic3rPE 1.42.0-beta2 on a non-main monitor (If that triggers any ideas).

Czappy, 2019-06-18:
Same issue for me. As soon as i move the slicer to the second monitor it crashes and isnt restarting again on that monitor. i have to turn it off to let the slicer start on the main monitor again to get it work again.

nkoeppel, 2019-06-23:
I'm having a seemingly related issue with my 1080ti, if i open PrusaSlicer and it doesn't open on my primary display it immediately crashes upon opening. If all of my displays mirror the primary display it will reset the display it opens on and resumes functioning correctly, I can then return my displays to extend mode and things will work properly unless I close PrusaSlicer on a display other than my primary display in which case it will fail to start the next time I try to open it.

bubnikv, 2019-06-24:
@Braddg I have a desktop with two NVIDIA 1080Ti graphics cards, driving three monitors.

How do you have these two cards configured? Do they act as a one or as two cards? I don't have personal experience with such a setup, but I believe the two cards could be interconnected with a special card to act as a single device.

bubnikv, 2019-06-24:
@nkoeppel Do you have just the single NVIDIA 1080Ti running on your system, or do you run on multiple graphics cards?

bubnikv, 2019-06-24:
@GeusB Is the on board GPU active or disabled?

I am asking many questions, because it may very well be, that we would have to set up a system to replicate those issues. It is virtually impossible to fix such an issue without being able to lay hands on the system.

bubnikv, 2019-06-24:
I suppose all these issues are on Windows only. Is anybody of you technical enough (a programmer?) to be able to compile and run slicer under a debugger?

nkoeppel, 2019-06-24:
I'm only running one graphics card

Braddg, 2019-06-25:
> @Braddg I have a desktop with two NVIDIA 1080Ti graphics cards, driving three monitors.
>
> How do you have these two cards configured? Do they act as a one or as two cards? I don't have personal [...]

I do not have the cards in SLI mode (single output).  

Ok, so per your request, I compiled and run under a debugger.  Programming is not my day job, so please forgive me if I didn't do this correctly.  When I open PrusaSlicer on a non-main monitor, it stops responding, so I break-all.  Attached is what I see.
[image]

bubnikv, 2019-06-25:
@Braddg Thanks a lot. This certainly gives a very useful hint. I cannot estimate whether it will help much, it may or may not. Now we shall google around.

bubnikv, 2019-06-25:
Would you guys please try the following build?
https://www.dropbox.com/s/ehzxcbygrqk3hry/PrusaSlicer-2.0.0%2B232-win64-g3d755e1bb-201906250722.zip?dl=0

It fixes a flickering issue on some NVidia cards. Let's keep our fingers
crossed that it helps at least to some of you.

Braddg, 2019-06-25:
The build did not work for me. 

This link has some interesting info:
https://caseymuratori.com/blog_0026

Are we creating the OpenGL render context on the correct HDC (monitor)?

Braddg, 2019-06-25:
I've noticed when placing a breakpoint on GetHDC() and stepping through, the HDC is defined when only the Console window is open on my main monitor.  The image below shows a comment, "HDC for this window, we keep it all the time".  So... if Prusa-Slicer-Console is defining the HDC then that will not work when rendering the GUI which is on a different monitor (which should be a different HDC)?

[image]

bubnikv, 2019-06-26:
Guys,

we need to sort out, who is able to run the Slic3rPE-1.41.3 and who is not.
That is, on which of your computers we broke something in PrusaSlicer
compared to Slic3rPE-1.41.3, and on which computers Slic3rPE-1.41.3 did not
work either.

We are not doing anything special about the OpenGL context to be monitor
specific. It works on all of our multi-monitor systems, but I admit that it
may not work on multi-graphic card setups. We need to dig further.

Braddg, 2019-06-29:
I was able to resolve my issue....

Stop the Nahimic service using task manager.  Somehow it is interfering with the rendering of the print bed.  When I turn it back on, my issue comes back.  

From what I can tell, both ASUS and MSI motherboards use the Nahimic service for audio.  For ASUS, I think it is installed with Sonic Studio.  I had to go into services and disable Nahimic from starting automatically.

Hope this works for everyone!

nkoeppel, 2019-06-29:
I do in fact have Sonic studio installed, if I have my issue again I'll try
that, sounds like the culprit.

nkoeppel, 2019-06-29:
Confirmed if I kill Sonic studio Prusa Slicer will start normally on any
display.

bubnikv, 2019-07-01:
@Braddg @nkoeppel thanks for heads up. This is hilarious. What would be the minimum / cheapest setup for us to buy to reproduce this issue? Can we install this Nahimic stuff and will slicer break on our computers?

bubnikv, 2019-07-01:
I have passed the following report to https://www.nahimic.com/solutions/#ContactForm

-----------------------------------------------

Hello.

We are having issues with dual screen setups and our PrusaSlicer application. Our application crashes, if your Nahimic service is running and one moves our PrusaSlicer app window from one screen to the other.

https://github.com/prusa3d/PrusaSlicer/issues/2522

It seems that your suite is causing havoc to other 3D graphics applications as well.

https://forum-en.msi.com/index.php?topic=284528.0
https://answers.unity.com/questions/1124406/strange-ui-editor-graphic-gltiches-in-unity3d-5-on.html?page=1&pageSize=5&sort=votes

Would you please help us to resolve the issue? Thank you.

St-Clair, 2019-07-24:
I have been having this issue for 6 months and only just bothered to check if I was the only one having the issue.

Just to add to the currently known problemwith Nahimic, their base version of sound drivers dont cause this issue (In my limited testing with other software), its something to do with the implementation that MSI and a few other resellers/repackagers use.

I would say it might be better to contact MSI/ASUS and the other repackers but this has been an issue for almost 4 years now, as an IT professional its disgusting to me that nothing is being done and sadly I dont think it will ever be fixed.

It may just be best to officially state that Prusa recommends uninstalling and/or stopping the tasks for Nahimic drivers and Sonic Studio if using Prusa Slicer on a multi-monitor setup.

breakthrulx, 2020-09-16:
Great thread guys. I have a recently purshased MSI Modern 15, using Prusa Slicer on secondary display, and suddenly today it refused to dipslay the plater when I opened the app.  If it wasnt for you guys I'd be thrust into a spiral of wiping, re-installing, and who knows what else. So...just to help those with the Nahimic app installed..... See the attached screenshot that shows the SoundTracker Engine that needs to be disabled. Notice the underlined red line. I guess that's their way of addressing the problem. So disabling the 'SoundTrackerEngine' worked. I was able to reopen Prusa Slicer no problem.
[image]

kocikdav, 2021-02-01:
Hello. I would like to ask users having troubles with Nahimic libraries to try this build. It should display warning dialog about NahimicOSD.dll linked.
https://www.dropbox.com/s/on0dwea75gfx4cc/PrusaSlicer-2.3.0%2B127-win64-gf939cac84-202102011345.zip?dl=0

We know that this is just a warning rather than a solution, but currently we have no way to stop loading this library as it comes linked statically when we start our process.

Also if you are aware of another libraries of same source please let us know their names. You can find it with ProcessExplorer tool.

bubnikv, 2021-07-30:
Warning about the Nahimic DLL being injected into PrusaSlicer will be rolled out with PrusaSlicer 2.4.0-alpha1. It is not a "fix" for the nahimic issue, but at least the user will be informed.

closed 2021-07-30 by bubnikv