prusa3d/PrusaSlicer #2524: Add hover over tool tips to tool bar functions

ktoms71, 2019-06-16:
### Version
 PrusaSlicer 2.0.0+

 
### Operating system type + version
MacOS 10.12.6 Sierra
### 3D printer brand / version + firmware version (if known)
MK3S with firmware version  3.7.1-2266

### Behavior
Hello,  I would like to request that hover over tool tips are added to the floating menus.  At least by adding the function name would help users of this software, which is great by the way.  The three floating toolbars within scope are shown in the attached screenshots. 
 
[image]
[image]

neophyl, 2019-06-17:
Possibly a Mac related issue as on Windows 7 I get them

[image]

bubnikv, 2019-08-01:
It is indeed an OSX specific wxWidgets issue, the tooltips not always work reliably.

closed 2026-09-29 by prusaslicer-bot[bot]