prusa3d/PrusaSlicer #2525: gcode output includes change to 1500C

uchobby, 2019-06-17:
### Version
2.0.0+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Windows 10 64B_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_Pursa MK3s FW: 3.7.1-2266_

### Behavior
* gcode output changes extruder temperature, early in the print, to 1500C.  The actual line in gcode is
M104 S1500 ; set temperature

* _Slice with Pursa Slicer a filamentTestObject.stl  This stl file has been used by me for months to test filament samples and has always printed fine. In this case I created a new filament type with a print temp of 240C and bed of 50c.  Other settings were likely basic for PLA_
  * _If this is a command-line slicing issue, include the options used_
* _PLA style print at a higher extruder temp (240C)_
* _MK3S stopped the print with a "Preheat Error" shortly after receiving the temp change to 1500. This is also a bug in the printer firmware. It should have indicated some other useful message._
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[FilamentTestObject.zip](https://github.com/prusa3d/PrusaSlicer/files/3295005/FilamentTestObject.zip)

lukasmatena, 2019-06-17:
This is what I see after opening your 3MF. I'm not sure I understand the problem.

[image]

chrisboer, 2019-06-17:
I have this issue after using the MMU profiles in Prusaslicer
[Brabantia_v2.0_0.15mm_PLA_MK3SMMU2S_1h6m.zip](https://github.com/prusa3d/PrusaSlicer/files/3295458/Brabantia_v2.0_0.15mm_PLA_MK3SMMU2S_1h6m.zip)

So in slicer I set PLA first layer temp 215 and other layers 215

M104 S215 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S215 ; wait for extruder temp

The printer lowers the temperature to PLA Prusa default 205 after first layer. Even when my other layer temp is set to 215. It is not storing my own PLA profile !!!! See attached .jpg
;AFTER_LAYER_CHANGE
;0.4
M104 S205 ; set temperature
[Brabantia_v2.0_0.15mm_PLA_MK3SMMU2S_1h6m.zip](https://github.com/prusa3d/PrusaSlicer/files/3295461/Brabantia_v2.0_0.15mm_PLA_MK3SMMU2S_1h6m.zip)

Why is this M104 S205 added to this GCODE when I did not ask for this to be put in... where can I remove these default.
As I can only find these so I am clueless
;AFTER_LAYER_CHANGE
;[layer_z]

chrisboer, 2019-06-17:
It now even get stranger, i use a new profile PLA Prusa Copy saved it with 
215 nozzle - 65 bed and now I get 255 and 60 in my gcode output ?
This is definitively a bug
[Brabantia_profile used v4.zip](https://github.com/prusa3d/PrusaSlicer/files/3295564/Brabantia_profile.used.v4.zip)

uchobby, 2019-06-17:
Maybe it was an operator in my case, I don't think I set the temp to 1500. But I'll agree that I must have.

I wonder if this kind of error should be caught someplace.  The printer could have said "temp out of range" maybe.  The "Preheat error" took some time to work through as documentation and the error string lead you down a bad path.

Is there a protection system in the printer for extream temperature?  I assume my printer would have been badly damaged if it had not failed due to the "Preheat error" until after the nozzle maxed out on the heater.

I will also try resetting the temps and slicing again, in case the temp gets changed by the process.

chrisboer, 2019-06-17:
@uchobby I don't think it is a user error I think there is an issue with profiles within Prusa Slicer as it don't saves the correct profile in the 3mf as well as exporting wrong parameters in the gcode

lukasmatena, 2019-06-17:
> I don't think I set the temp to 1500. But I'll agree that I must have.

Most likely you typed 2400 instead of 240 and the field capped the value with a maximum that it can accept. Which is 1500 (see also https://github.com/prusa3d/PrusaSlicer/issues/629).

> Is there a protection system in the printer for extream temperature?

I would guess that's what triggered the error. What the printer should do if it receives such request should be discussed in our firmware github repository, not here.

> I will also try resetting the temps and slicing again,

Please, close the issue if the problem disappears.

@chrisboer Would you please stop mixing several issues together? Your problem has nothing to do with what's being discussed here. It is also likely a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2204 so that might be a better place to go.

uchobby, 2019-06-18:
I have confirmed that entering 2400 was saved as 1500 so the speculation above seems the likely explanation.

closed 2019-06-18 by uchobby