prusa3d/PrusaSlicer #2527: Feature request

lokus77, 2019-06-17:
Would be nice (and maybe unique feature, does not exist in any slicer) if the user could **paint in 3d view the seam position** (across Z ). So user can decide where goes seam in each part of a model. Would be something like variable layer height editor. Not sure if is easy to implement.

Excelent Slicer, nice job prusa development team!

maximlevitsky, 2019-06-17:
I would also like to have that feature.

armsofilum, 2019-07-22:
This is something I'd REALLY like to see added. For production-ready parts, I'd argue that it's a necessary feature. A few thoughts:

Cura's option to make the z-seam placement relative to the selected model is very helpful when working with several of the same model on the build plate. Cura, however, does not allow z-seam placement to be changed by layer. S3D on the other hand, forces explicit coordinates for the z-seam for each process/model. That level of granular control is appreciated, but the coordinates aren't relative to the model (they reference specific spots on the plate) meaning that working with several models can be clunky and moving a model means resetting the coordinates. Unlike Cura, S3D allows this placement to be modified at any layer.

The optimal solution would be to incorporate a setting that introduces Cura-like options to define the seam coordinates as relative to the model's center (this is the most important aspect), while also adding the ability to change that setting depending on the layer (a nice to have). I don't personally care if the seam is painted (in fact, this adds complexity), I would just like more control over placement.

electronicamystica, 2019-10-24:
As a workaround I currently just use "Seam position: rear" and rotate the object so that the seam aligns with the rear position. Works fine with round objects or if the seam has to be in a corner of some angular object. with a cube i have to rotate it by 20 degrees, 15 will just place the seam off center of the side facing backwards.
Seems someone else worked out a function like that earlier: https://github.com/prusa3d/PrusaSlicer/pull/728

n8bot, 2020-06-22:
This is a fantastic idea.

The slicer IceSL uses painting with a brush as a way to locate seam placement. They also allow the user to paint places to block supports.

Now that PrusaSlicer is getting paint-to-locate supports, it could be a good time to think about adding paint-to-locate seam.

lukasmatena, 2020-06-23:
> Now that PrusaSlicer is getting paint-to-locate supports, it could be a good time to think about adding paint-to-locate seam.

We had exactly this idea. This will be addressed soon. 
This is also a duplicate of #946.

lukasmatena, 2020-10-16:
PrusaSlicer 2.3.0 introduces a paint-on seam. The alpha will be out in a few days, we'd be glad if you could test it so all bugs can be fixed.

Closing the issue.

closed 2020-10-16 by lukasmatena