prusa3d/PrusaSlicer #2529: Modifier: Layer Height not working

GWdd, 2019-06-17:
### Version
2.0.0 Win64

### Operating system type + version
Windows 7 Pro

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3

### Behavior
Part Modifier does not change layer height.

Part is simple inverted box, layer height set to 0.15 mm via profiles, with 2 perimeters.

In the modifier, I have set perimeters to 3 and layer height to 0.1.

Result is area within modifier has 3 perimeters, but unchanged layer height.

[image]
[image]
[image]

_Is this a new feature request?_
Nope

#### Project File (.3MF) where problem occurs
[Modifier.zip](https://github.com/prusa3d/PrusaSlicer/files/3298500/Modifier.zip)

Qwetek, 2019-06-19:
Same problem, tried RC2 still there. Other modifies work just fine.

YuSanka, 2019-06-20:
There shouldn't be "layer height" option for modifiers at all. It's a bug. It shouldn't be possible to add "layer height" using "Layers and Perimeters" context menu item.
This bug is already fixed with [commit](https://github.com/prusa3d/PrusaSlicer/commit/67ed89c2405a242370066993c16e26fa26fdb6c4) and will be a part of the upcoming release. 

And yes, the possibility of editing different range of layers will be implemented too. We are currently working on it.

bubnikv, 2019-08-01:
Layer height per layer height range will be implemented in the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-01 by bubnikv