prusa3d/PrusaSlicer #2530: Auto Arrange can't be disabled

GWdd, 2019-06-17:
### Version
_Version of PrusaSlicer used goes here_
2.0.0 win64

### Operating system type + version
Win 7 Pro x64 

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3

### Behavior
Adding parts (parts that are to be stacked) are being autoarranged even after manually setting the part positions.

Add a part, then add a part to that part, and the first part is autoarranged instead of being positioned relative to the original part.  Add another part and things move again.

 
[image]

_Is this a new feature request?_
Nope

#### Project File (.3MF) where problem occurs
[AdddPart.zip](https://github.com/prusa3d/PrusaSlicer/files/3298543/AdddPart.zip)

Desired stacked positions ...
[image]

neophyl, 2019-06-18:
Is this what you are trying to do ? (I might not have them stacked perfectly but you get the gist)

[image]

[case0.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3300440/case0.3mf.zip)

bubnikv, 2019-06-18:
That is, you should load the multiple objects as volumes of a single Slicer
object.

GWdd, 2019-06-18:
Bub - Can you better define what you mean by "volumes of a single object" ??? 

Neo - yes, trying to find a way to change layer height as the stack build.  As in a discrete jump rather than a smoothed jump: section1 0.20 mm, section2 0.05 mm, section3 0.10 mm.  All the tools indicate it should be possible, but it isn't.

I agree the offset is likely the root cause in my example, but the ability to disable auto-arrange is still needed.  Disabling auto-center doesn't work here - something added that offset after I cut the part in Plicer.

bubnikv, 2019-08-01:
No auto-centering is going on, the objects are misaligned inside the 3mf. Closing.

closed 2019-08-01 by bubnikv