prusa3d/PrusaSlicer #2531: replacing supports on first layers

userosos, 2019-06-17:
### Version
Version: 2.0.0+win64
Build: PrusaSlicer-2.0.0+win64-201905201652
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build)

Hi! Can you add on first layer skirt near the part of model replacing support. I think that it will have a better effect on the adhesion of the model to the hot bed.

I want that this support (on first layers) replaced to skirt.

[image]

[EBA_01.00.010_basement.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3298686/EBA_01.00.010_basement.3mf.zip)

ManuGithubSteam, 2019-06-24:
Hi
AFAIK you can just use the skirt setting and also supports. Then it will make a skirt and build supports.

bubnikv, 2021-02-24:
Fixed with 055d2321e7d0b5727e1933a8c64ef6074bc2956
Supports will be printed over brim.

closed 2021-02-24 by bubnikv