prusa3d/PrusaSlicer #2533: Top fill pattern isn't used in spiral vase mode

jurriaan, 2019-06-18:
### Version
PrusaSlicer 2.0.0

### Operating system type + version
Linux 5.1.6-zen1-1-zen

### 3D printer brand / version + firmware version (if known)
Prusa MK3S on latest firmware (3.7.1?)

### Behavior
The top infill pattern is set to Archimedian Chords but it looks like rectilinear is being used for the top of the infill of the object I want to print.

This only happens in Spiral vase mode.

#### Project File (.3MF) where problem occurs
Created a demo of this issue with a simple box:

[box_vase_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/3300994/box_vase_bug.zip)

supermerill, 2019-07-05:
spiral vase mode has no infill...

or i missed something?

jurriaan, 2019-07-05:
@supermerill there are a few infill layers on the bottom of the vase, and I would expect that the top infill setting would affect the top layer (i.e. the top layer of the vase's bottom) just as the bottom infill setting affects the bottom.

The current behavior is that it ignores the setting and defaults to rectilinear.

supermerill, 2019-07-06:
oh.
vase mode is a hack where there are no top nor infill and the perimeter can be 'spiralled'.
So no top layer, because there aren't (for now).

jurriaan, 2019-07-17:
@supermerill that option (infill pattern) is disabled in PrusaSlicer 2.0, so that's not really possible. Also, I do think that using the same pattern throughout the bottom will affect stability. Ideally I want to adjust these top and bottom layers for aesthetic purposes, while keeping the inner infill rectilinear for stability and in case of a real vase, so it doesn't leak.

DrLex0, 2019-07-17:
I also encountered this yesterday. It makes sense indeed because of the way spiral vase works. My workaround was to change all infill settings to what I wanted for the top infill, which of course made the whole print much more time-consuming than it needed to be, and as @jurriaan says, those settings may not be desirable for inner infill layers. I would also like to see that bit of effort being spent on treating the last solid layer in vase mode as a top layer.

bubnikv, 2020-02-08:
Spiral vase improvements and bugfixes.
Fixes Connecting / expanding Bottom Layers to Vase Perimeter #253
Fixes Slicing error in vase mode #452
Fixes Slicing Issue (Vase Mode, 0.6mm dmr nozzle) #1887
Fixes Top fill pattern isn't used in spiral vase mode #2533
Fixes Cisar's vase doesn't slice correctly, creates artefacts #3595

When the model is sliced, all the contours are newly oriented counter-clockwise (even holes), merged and then only the largest area contour is retained. In perimeter generator, if the largest contour splits into multiple perimeters, newly only the largest area perimeter is retained in spiral vase mode. These two changes solve #3595 and similar.

The infill is newly calculated only for the bottom solid layers if the spiral vase mode is active (removes various unwanted infill along the vase walls), and the last bottom solid layer is switched to a top solid pattern (solves #2533).

The thin walls are newly enforced to be disabled in spiral vase mode, and the "ensure vertical shell wall" is enforced in spiral vase mode to extend the bottom of the vase to the vase hull (fixes #253).

The changes will be released with PrusaSlicer 2.2.0-alpha4. Closing.

bubnikv, 2020-02-08:
Fixed with 4e11552da980f07216f1720f67f539703ed90ed1

Spiral vase improvements and bugfixes.
Fixes Connecting / expanding Bottom Layers to Vase Perimeter #253
Fixes Slicing error in vase mode #452
Fixes Slicing Issue (Vase Mode, 0.6mm dmr nozzle) #1887
Fixes Top fill pattern isn't used in spiral vase mode #2533
Fixes Cisar's vase doesn't slice correctly, creates artefacts #3595

When the model is sliced, all the contours are newly oriented counter-clockwise (even holes), merged and then only the largest area contour is retained. In perimeter generator, if the largest contour splits into multiple perimeters, newly only the largest area perimeter is retained in spiral vase mode. These two changes solve #3595 and similar.

The infill is newly calculated only for the bottom solid layers if the spiral vase mode is active (removes various unwanted infill along the vase walls), and the last bottom solid layer is switched to a top solid pattern (solves #2533).

The thin walls are newly enforced to be disabled in spiral vase mode, and the "ensure vertical shell wall" is enforced in spiral vase mode to extend the bottom of the vase to the vase hull (fixes #253).

The changes will be released with PrusaSlicer 2.2.0-alpha4. Closing.

closed 2020-02-08 by bubnikv

jurriaan, 2020-02-08:
@bubnikv It seems that your commit broke some stuff:

```
In file included from /home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/src/libslic3r/PrintBase.hpp:14,
                 from /home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/src/libslic3r/SLAPrint.hpp:5,
                 from /home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/tests/sla_print/sla_test_utils.hpp:13,
                 from /home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/tests/sla_print/sla_print_tests.cpp:5:
/usr/include/tbb/mutex.h:21:140: note: #pragma message: TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.
   21 | #pragma message("TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                                            ^
/home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/tests/sla_print/sla_print_tests.cpp: In function ‘void ____C_A_T_C_H____T_E_S_T____2()’:
/home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/tests/sla_print/sla_print_tests.cpp:57:58: error: no matching function for call to ‘Slic3r::TriangleMeshSlicer::slice(std::vector<float>&, const float&, std::vector<std::vector<Slic3r::ExPolygon> >*, ____C_A_T_C_H____T_E_S_T____2()::<lambda()>)’
   57 |     slicer.slice(slicegrid, CLOSING_RADIUS, &slices, []{});
      |                                                          ^
In file included from /home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/tests/test_utils.hpp:4,
                 from /home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/tests/sla_print/sla_test_utils.hpp:5,
                 from /home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/tests/sla_print/sla_print_tests.cpp:5:
/home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/src/libslic3r/TriangleMesh.hpp:182:10: note: candidate: ‘void Slic3r::TriangleMeshSlicer::slice(const std::vector<float>&, Slic3r::SlicingMode, std::vector<std::vector<Slic3r::Polygon> >*, Slic3r::TriangleMeshSlicer::throw_on_cancel_callback_type) const’
  182 |     void slice(const std::vector<float> &z, SlicingMode mode, std::vector<Polygons>* layers, throw_on_cancel_callback_type throw_on_cancel) const;
      |          ^~~~~
/home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/src/libslic3r/TriangleMesh.hpp:182:57: note:   no known conversion for argument 2 from ‘const float’ to ‘Slic3r::SlicingMode’
  182 |     void slice(const std::vector<float> &z, SlicingMode mode, std::vector<Polygons>* layers, throw_on_cancel_callback_type throw_on_cancel) const;
      |                                             ~~~~~~~~~~~~^~~~
/home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/src/libslic3r/TriangleMesh.hpp:183:10: note: candidate: ‘void Slic3r::TriangleMeshSlicer::slice(const std::vector<float>&, Slic3r::SlicingMode, float, std::vector<std::vector<Slic3r::ExPolygon> >*, Slic3r::TriangleMeshSlicer::throw_on_cancel_callback_type) const’
  183 |     void slice(const std::vector<float> &z, SlicingMode mode, const float closing_radius, std::vector<ExPolygons>* layers, throw_on_cancel_callback_type throw_@bubnikv It seems that your commit broke some stuff:

```In file included from /home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/src/libslic3r/PrintBase.hpp:14,
                 from /home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/src/libslic3r/SLAPrint.hpp:5,
                 from /home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/tests/sla_print/sla_test_utils.hpp:13,
                 from /home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/tests/sla_print/sla_print_tests.cpp:5:
/usr/include/tbb/mutex.h:21:140: note: #pragma message: TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.
   21 | #pragma message("TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                                            ^
/home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/tests/sla_print/sla_print_tests.cpp: In function ‘void ____C_A_T_C_H____T_E_S_T____2()’:
/home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/tests/sla_print/sla_print_tests.cpp:57:58: error: no matching function for call to ‘Slic3r::TriangleMeshSlicer::slice(std::vector<float>&, const float&, std::vector<std::vector<Slic3r::ExPolygon> >*, ____C_A_T_C_H____T_E_S_T____2()::<lambda()>)’
   57 |     slicer.slice(slicegrid, CLOSING_RADIUS, &slices, []{});
      |                                                          ^
In file included from /home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/tests/test_utils.hpp:4,
                 from /home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/tests/sla_print/sla_test_utils.hpp:5,
                 from /home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/tests/sla_print/sla_print_tests.cpp:5:on_cancel) const;
      |          ^~~~~
/home/jurriaan/.cache/yay/prusa-slicer-git/src/PrusaSlicer/src/libslic3r/TriangleMesh.hpp:183:10: note:   candidate expects 5 arguments, 4 provided
make[2]: *** [tests/sla_print/CMakeFiles/sla_print_tests.dir/build.make:76: tests/sla_print/CMakeFiles/sla_print_tests.dir/sla_print_tests.cpp.o] Error 1
```

bubnikv, 2020-02-09:
fixed with e7d2fcf0be7a524eb8bf40dd2964539534934e96