prusa3d/PrusaSlicer #2534: Printer do not preheat - StartingGcode not working?

Czappy, 2019-06-18:
Prusa Slicer
Version 2.0.0
Firmware: Marlin 2.0.x Bugfix
Printer: Alfawise U20 at SKR 1.3

My problem is that whatever startcode i use the printer is immediatley starting to print after the Code is done. So it´s not preheating the Bed nor the nozzle.

I took my code from S3D where it works.
Maybe not in PrusaSlicer?

M190 S60 ; Bett 60 Grad Heizen,
G28 ; Alle Achsen Homen,
G29 P5 ; 5x5 Bltouch Leveln,
M109 S205 ; Düse PLA 205 Grad heizen,
G92 E0 ; Extruder Position setzen,
G1 Z2.0 F3000 ; Z Achse 2mm anheben um das Bett nicht zu zerkratzen,
G1 X299.1 Y20 Z0.3 F5000.0 ; Fahre zum Startpunkt,
G1 X299.1 Y280.0 Z0.3 F1500.0 E15 ; Drucke Linie 1,
G1 X298.8 Y280.0 Z0.3 F5000.0 ; Bisschen Rüber,
G1 X298.8 Y20 Z0.3 F1500.0 E30 ; Drucke Linie 2,
G1 E27 F3000 ; Filament Einzug 3mm,
G92 E0 ; Extruder Position,
G1 Z5.0 F3000 ; Z anheben um Druckbett zerkratzen zu vermeiden,
M117 Print…!,

lukasmatena, 2019-07-22:
I don't understand. The gcode you sent is from S3D? If so, I don't understand its purpose. Please, enclose a 3MF with complete PrusaSlicer configuration and ideally the full GCode so we can see what's wrong. Thanks.

lukasmatena, 2019-11-28:
No response, closing. No other issues like this reported.

closed 2019-11-28 by lukasmatena