prusa3d/PrusaSlicer #2535: Modifications

artdezigner, 2019-06-18:
Please add Russian language and USB printing for custom printers - it is very necessary

bubnikv, 2019-06-19:
If Russian is very extremely necessary, then please provide the
translations.

artdezigner, 2019-06-19:
Ha, interesting approach to the creation of the program=) Probably about printing via USB is also offer do it yourself? When creating Cura programmers themselves coped with the languages, if You are going to make a competitor Cura, then try to make PrusaSlicer at least with the same functionality. Now use PrusaClicer uncomfortable.

dixiony, 2019-06-21:
Programmers do not owe you anything.
If you want the program to be in Russian, then read about it here: https://3dtoday.ru/blogs/tarasik99/slic3rpe-russian-localization/
If you want to print on USB, then there is the program Repeatier-Host.
---
RUS: думаю, я понятно написал. Если хотите видеть русский перевод, то присоединяйтесь к команде переводчиков, к тому же перевод для предыдущей версии Slic3R PE уже есть в наличие.
Если хотите печатать через USB, то используйте Repetier Host - лично не вижу никакой проблемы и если нужно, то всегда так и пользуюсь.

bubnikv, 2019-06-24:
@artdezigner 
> When creating Cura programmers themselves coped with the languages, if You are going to make a competitor Cura, then try to make PrusaSlicer at least with the same functionality. Now use PrusaClicer uncomfortable.

Спасибо за ваше предложение.

bubnikv, 2019-06-24:
@dixiony Would you please guys finish the translation and prepare a pull request? We will be more than happy to integrate it. Thank you.

bubnikv, 2019-08-01:
We do not plan to translate to Russian ourselves, and the USB printing is mentioned elsewhere. Clsoing.

closed 2019-08-01 by bubnikv

artdezigner, 2019-08-01:
where's the answer to the question?