prusa3d/PrusaSlicer #2536: Arranging of certain objects causes infinite loop.

ryanneufeld, 2019-06-18:
### Version
Tested in both 1.43 latest and 2.x latest.

### Operating system type + version
macOS 10.13.6 (17G7024)

### 3D printer brand / version + firmware version (if known)
Original MK3

### Behavior
Load the attached/linked STL. with at least two copies/instances (either way will cause the issue)

Click the auto arrange button, and witness the beachball and unresponsive application.

#### Project File (.3MF) where problem occurs
[Battery_Adapter_AA-C.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3303504/Battery_Adapter_AA-C.3mf.zip)

tamasmeszaros, 2019-06-19:
@ryanneufeld, thank you for the feedback, I will look into this.

bubnikv, 2019-08-01:
This issue will be fixed by the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-01 by bubnikv