prusa3d/PrusaSlicer #2537: Bridge Infill rendering and slicing

GWdd, 2019-06-19:
### Version
2.0.0+Win64

### Operating system type + version
Windows 7 Pro x64

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3 v3.7.1

### Behavior
When slicing holes in objects, the top of the hole seems to have slice issues where excess material is being extruded.  Some of this may be rendering of bridge infill (bridge infill is rendered twice the thickness as other layers.)   The printer also lays down excessive plastic, as if the slice does expect thicker extrusion. 

[image]

The Bridge Infill also slices at unexpected angles.  Set 0 degrees and get 0 and 10.
[image]

This all leads to a less than desirable hole, with visible over extrusion at the top.
[image]

It would be nice if holes on a vertical surface were sliced with some level of expectation the printed hole will be as round as the layers permit.

#### Project File (.3MF) where problem occurs
[Hole.zip](https://github.com/prusa3d/PrusaSlicer/files/3304424/Hole.zip)

supermerill, 2019-07-05:
To explain a bit more what is going on:

The slicer over-extrude bridge: when it slice, it thinks that the layer height will be the height of every extrusion. When the surface is filled with bridged extrusion, the height of the extrusion is changed to layer width (as the extrusion can only be round), so it's over-extruded every time (too low).

The angle of bridges is computed by a (deficient) algorithm (that i fail to correct for now), it shouldn't take into account your setting.

Another source of over-extrusion is that the bridged surface go very deep into solid areas, and you can't modify that value (for now) in prusaslicer, and it doesn't change the extrusion type to solid in these areas.

Reocin, 2021-04-22:
To add more information to this problem. I also found a similar problem especially with variable layer high. Here are some pictures.
I already opened another issue. 
https://github.com/prusa3d/PrusaSlicer/issues/6414

[image]

[image]

closed 2026-09-29 by prusaslicer-bot[bot]