prusa3d/PrusaSlicer #2538: Feature Request: Adding custom settings to filaments.

AmyTheCute, 2019-06-19:
I'd love a feature where it lets me decide what settings go in the per-filament department. let me share an example.

lets say I want to add different settings for my TPU filament. the speeds need to go down to 35/30mm/s and be the same value (I set every single speed to the same value for TPU, it helps a lot, so this way there's less sudden speed changes, pressure problems). But I print most other things at 45 and maybe different speeds for infill etc.

**what I have to do:**
Create a new printer profile called Printer - TPU
remember to not only change my filament but change my printer settings separately.

**What I'd like to do instead:**
Add speed as a per filament settings so therefore when I choose TPU it changes the settings to. maybe as a percentage of the base speed etc. doesn't really matter much to me tho!

another example: retraction

**what I have to do:**
Create seperate profiles for ABS/TPU/PLA in the printer section.
**what I'd like to do:**
add retractions to the per filament section and make it be filament dependant.

this is a very small quality of life change but I thought I just mention it :) If letting people add options to the filament section is too hard (coding wise) maybe there's just a few mroe commonly changed things you could add? like retractions print speed percentage for all premiers etc.

the issue with different print profiles for each filament is lets say I change my nozzle to 0.3 and change it in my PLA profile. I can't tell you how many times I've forgoten to go trough all my profiles to change it in every single one too. then it's just needless confusion about why my TPU prints are coming out weird.

lukasmatena, 2019-06-19:
@AmyTheCute This seems to be a duplicate of #480. If so, could you close this issue and discuss it there? (The issue was also brought up in #1556 and #844.)

AmyTheCute, 2019-06-19:
> @AmyTheCute This seems to be a duplicate of #480. If so, could you close this issue and discuss it there? (The issue was also brought up in #1556 and #844.)

Oh I"m sorry! I didn't see those. Although my feature request is different I realized we only miss those two settings really. I hope the developers add them soon :)

closed 2019-06-19 by AmyTheCute