prusa3d/PrusaSlicer #2539: Scale can not be unlocked in Expert mode

davzarek, 2019-06-19:
### Version
_Version of PrusaSlicer used goes here_
_Use `About->About PrusaSlicer` for release versions_
2.0.0+win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
win64 - windows 10 pro

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
prusa mk3s

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
Scale can not be unlocked in Expert mode
Clicking the orange scale icon
It will not unlock

_Is this a new feature request?_
No, it is a bug

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
ALL .3mp files

YuSanka, 2019-06-20:
Thanks for your report. Unfortunately, we can't reproduce this issue. Could you please provide us a model file on which the issue occurs. A short video demonstrating the issue would be really great.

davzarek, 2019-06-20:
Its the 3D models you provided for the MMU2S, so just download them, nr 2
to be exact.

If that not works I can make a video!

Best
David

YuSanka, 2019-06-20:
Still nothing ( Everything works as it's expected. Is it possible, that you have multiple selection on right panel?

davzarek, 2019-06-20:
Hey!

Thats correct - I have multiple objects selected - So I can not scale
several objects at once?

That would be VERY handy - especially in Z-axis to adjust height of objects.

Best Regards
David Radszuweit

Project Manager
Six Sigma Black Belt
BFT | MFT
M: SWE: +46 70 6288887
E: davidrad@gmail.com
www.davidr.eu

YuSanka, 2019-06-20:
Thank you for your suggestions. But for this moment you can scale several objects at once only in uniform scaling mode.

davzarek, 2019-06-20:
OK!

That helped A LOT!

Problem is that its not intuitive - and a small "I" info box about
functions that are so crucial, could be useful!

All solved and you should make a video series for using the slicer!
Great marketing and only 500 USD to ask some forum guru to do it for the
prusa brand!

Best Regards
David Radszuweit

Project Manager
Six Sigma Black Belt
BFT | MFT
M: SWE: +46 70 6288887
E: davidrad@gmail.com
www.davidr.eu

bubnikv, 2019-08-01:
@YuSanka added a tooltip to the non-uniform scaling lock button informing, that the non-uniform scaling is not possible for group selection. This change will be part of the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-01 by bubnikv