prusa3d/PrusaSlicer #2540: [slicing bug] Some layers are double-printed or repeated

MrFlippant, 2019-06-19:
### Version
v2.0.0+win64

### Operating system type + version
Windows 10 Home, up to date

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3s with MMU2s and current release firmware on both.
Mk3s has been modified with new x-ends, y-axis tension adjustment and x carriage. MMU2 has been modified with wide bore selector with magnet mod and underslung control panel.
Firmware: Mk3s - 3.7.1-2266; MMU2 - 1.0.5-297

### Behavior
Some parts of the model are having the same layer printed at least two times in succession. There is no indication of this in the slicer preview that I could see. When the part showed signs of overextrusion, I watched it print and saw that the same area was being printed multiple times in a row at the same layer height. This is on a waste object, so it should not have any "doubled-up" as on the primary object when printing multiple colors prior to changing filament, but even that is two different layers at two layer heights. This problem is the same layer, at the same layer height, being printed more than once, possibly more than twice.

#### Project File (.3MF) where problem occurs
Attachment includes 3mf, gcode, and photos.
[Fidgets.zip](https://github.com/prusa3d/PrusaSlicer/files/3306681/Fidgets.zip)
[image]
[image]
[image]

standarch, 2019-06-29:
Agreed, can't see a problem in the preview of the sliced object using PrusaSlicer2.0.0 on linux64 and Mk3s printer setting. Looks like either G code not generated properly, or some problem with the printer interpreting the G code.
OK
The G code is not being generated correctly. I used an online viewer of the gcode path. Notice that there is dark section of the photo.
[image]

standarch, 2019-06-29:
The dark section is I think caused by repeated traces causing the image to be darker.

standarch, 2019-06-29:
I think possibly someone has overlayed two parts in exactly the same location.

There are 3 identical groups of parts.  It could be possible to delete one of these, and then duplicate one of the remaining two copies.

e.g. delete the selected group in the picture and copy one of the other groups which are ok.
[image]

standarch, 2019-06-29:
Although maybe this causes the printed part to be the wrong colour? The subparts of the group can be selected in Prusa slicer, and the extruder (i.e. the part colour) changed.  Its in the expert settings.
[image]

bubnikv, 2019-08-01:
@MrFlippant  it seems that @standarch found out that you have multiple overlapping objects in your scene. Does that finding solve your issue? Could you please close this issue?

MrFlippant, 2019-08-01:
Except there were not. The problem is not in my scene, but in how that scene was sliced. Maybe it was an anomaly, which is why I included the GCode for you guys to look at. Since the issue has not been addressed, I don't feel that it should be closed.

I was not looking for a solution to what happened, as I changed the scene and re-sliced and printed just fine. The point of this was to provide information about a slicing error, which I assume is related to a bug in the slicer, and that those in control of the software would want to know about it.

bubnikv, 2019-08-01:
It really looks like the G-code attached contains duplicate G-code at the marked location.

[image]

bubnikv, 2019-08-01:
Our current PrusaSlicer 2.1.0-alpha0 crashes on this code. There is something wrong going on.
@lukasmatena please look into it.

MrFlippant, 2019-08-01:
I do not argue that point. The point I'm making is that I did not edit the code. That's how it came out of Prusa Slicer, therefore there is a bug in the slicer, either in the slicing process, or the export/saving process.

standarch, 2019-08-01:
Something strange with the combination of 3mf file and slicer.  I can't delete certain components in slicer.  It may have lost track of certain components.
Version:
PrusaSlicer-2.0.0+linux64-201905201652

I guess slicer is not reading the 3mf file correctly.  Whether thats because the file is corrupt or slicer can't read very well, I don't know. The gcode is generated incorrectly, but I don't think the gcode generation is the main problem.

lukasmatena, 2019-08-01:
@MrFlippant We are thankful that you reported it, it really seems to be a bug in PrusaSlicer and we will do our best to fix it. Your scene appears to be fine.

@bubnikv The 3MF above slices to the end on my PC. There is no crash, no assert trips in debug mode, ASAN is silent. Whether or not there are duplicate extrusions I'll inspect at another time.

@standarch Could you please be more specific about what you are unable to do with the 3MF? I see nothing suspicious, but it could lead us to the cause of the problem.

Duplicate extrusions would suggest that the culprit is wipe into infill/object feature (which is enabled in that project). But it would be the first reported bug in that feature ever and I'd be surprised to see it surface now, after more than a year since it was implemented. Such an error is impossible to miss and would make anyone angry and eager to share. I'll look into it.

MrFlippant, 2019-08-01:
Well, I do understand that if it can't be replicated, it probably can't be fixed, and we can chalk it up to a fluke of my PC. It hasn't happened to me since this one. If it happens again, I'll post about it again and try to remember anything that might have been different to my normal activities.

@lukasmatena et al, I'll close this when next requested.

bubnikv, 2019-08-02:
@lukasmatena I am getting a crash when the file gets sliced for the second
time. I hope I can show you.

standarch, 2019-08-02:
In the first image there is a brown object.  This object has no name in the list.  If one selects it and presses the rubbish bin icon (delete all objects), the second image results.  Although an object still remains after delete all, there is still no object name in the list. Also the rubbish bin icon is now greyed out, and the remaining object can not be deleted.

PrusaSlicer-2.0.0+linux64-201905201652

[image]
[image]

lukasmatena, 2019-08-03:
@standarch That's wipe tower, which is there on purpose and can be turned off in Print Settings->Multiple extruders. This is the intended behavior.

bubnikv, 2019-08-05:
> I was not looking for a solution to what happened, as I changed the scene and re-sliced and printed just fine. The point of this was to provide information about a slicing error, which I assume is related to a bug in the slicer, and that those in control of the software would want to know about it.

I guess there either was some invalid memory access for some reason (which may have been fixed on the master code already), or you did change some parameters after the first slicing, and this change of parameters did not invalidate some of the back-end slicing data, that should have been invalidated (slicer tries to cache as much as possible to speed up the slicing, but that is an error prone process).

lukasmatena, 2019-09-04:
I finally managed to provoke this error in different scenario. It sometimes appears when you slice, switch extruders of the objects and slice again (it requires some specific switches). Looks like something is not invalidated when it should be.

Now that I can reproduce it, I should be able to fix it.

MrFlippant, 2019-09-04:
Yay! It's entirely likely that I did exactly that.

coolhoun, 2020-02-19:
I can also confirm this behavior in version 2.1.1 and 2.2.0 alpha 4, both using on MacOS.  The first layer prints twice, in certain circumstances.  I've tried different objects, different positioning, and (as shared) different versions with the same effect.  My setup is a custom machine, using 3 extruders and a Diamond mixing Hotend.  

I have 8 virtual extruders setup, I'm not as knowledgable as the prior posters in this thread in how to describe the issue.  But here is a recent file which I sliced and exhibits the issue.
[Five Racing Frogs.gcode.txt](https://github.com/prusa3d/PrusaSlicer/files/4223163/Five.Racing.Frogs.gcode.txt)

coolhoun, 2020-02-19:
@lukasmatena What information can I provide to assist you in rooting out this bug?

I have reproduced this slicing bug with two different STL files (www.thingiverse.com/thing:2659257 and www.thingiverse.com/thing:2429838, so far).   

My slicing issue hasn't gone away as easily as others, it's been pretty persistent, @standarch  method of using gcode.ws to inspect has been very helpful, saves me from actually printing the file over, and over..

coolhoun, 2020-02-19:
[Five frogs 2.gcode.txt](https://github.com/prusa3d/PrusaSlicer/files/4223388/Five.frogs.2.gcode.txt)
 Another example

coolhoun, 2020-02-19:
Another...Tried like 10 of these paddleboat, eventually I just let it lay down the first layer twice, and after that all seemed fine...
[PaddleBoat v2.gcode.txt](https://github.com/prusa3d/PrusaSlicer/files/4223396/PaddleBoat.v2.gcode.txt)

coolhoun, 2020-02-19:
My Prusa Slicer config file. (should be INI)
[PrusaSliver 2.2 config.txt](https://github.com/prusa3d/PrusaSlicer/files/4223400/PrusaSliver.2.2.config.txt)

coolhoun, 2020-02-19:
@lukasmatena I also used the "wipe to object" setting on both of these slices.    Also, trying with your Sep 4th comment in mind, I closed Prusa Slicer, recreated the scene, and ONLY sliced once, then exported code.  The problem persisted (that is the attached Five frogs 2.gcode above).

coolhoun, 2020-02-19:
I disabled the "wipe to Object" and "wipe to infill" on the 5th frog - used gCode analysis and -poof, fixed.  I went immediately back to Prusa Slicer, flipped those switches to 'on' again, so that 5th frog would be the wipe object, and poof, double printing of first layer (and in this case, I think there is actually duplication of more than just the first layer, I can upload one of these codes if need be).

@lukasmatena thank you so much for locating this error and identifying the setting related!

lukasmatena, 2020-02-19:
@coolhoun Thanks. I would just like to say that there is another bug in 2.2.0-alpha4, that might trigger similar behaviour (https://github.com/prusa3d/PrusaSlicer/issues/3665). That one was not present in 2.1.1 and is already fixed for 2.2.0-beta.

bubnikv, 2020-02-19:
@coolhoun  could you please do us a favor?

Would you attach as small as possible 3mf with as short as possible sequence of steps to reproduce the issue? With such data we are half way fixing the issue. Thank you.

coolhoun, 2020-02-19:
@bubnikv Of course.  Here are the steps I took to reproduce the issue:
- Open Prusa Slicer 2.2.0 alpha 4 (I am running MacOS 10.14.6)
- Select “add part”, select Yin.stl and Yang.stl, and select ‘yes’ to multi-part dialog box.
- Repeat to add another Yin Yang coin
- Orient objects
- Set the 2nd Yin Yang Wipe object flags, both of them
- Slice once, (Wipe tower was outside bounds, relocate, sliced again)
- Uploaded to gcode.ws and confirmed that the first layer doubles, as well as second layer, possibly others…

[Yin-Slc3rPE2.2.gcode.txt](https://github.com/prusa3d/PrusaSlicer/files/4226778/Yin-Slc3rPE2.2.gcode.txt)
[Yin.3mf.txt](https://github.com/prusa3d/PrusaSlicer/files/4226779/Yin.3mf.txt)
[Yin.zip.amf.txt](https://github.com/prusa3d/PrusaSlicer/files/4226780/Yin.zip.amf.txt)

[Yang.stl.txt](https://github.com/prusa3d/PrusaSlicer/files/4226785/Yang.stl.txt)
[Yin.stl.txt](https://github.com/prusa3d/PrusaSlicer/files/4226787/Yin.stl.txt)

bubnikv, 2020-02-20:
I don't know what happens in your case, but when I load your 3mf, I see the following:

[image]

It looks like you have your model imported multiple times overlapping on the print bed, there is no wonder that slicer then extrudes one copy over the other.

coolhoun, 2020-02-20:
Strange, I don't see that when I open the same 3mf on my computer (in either Slic3r 1.42 alpha or PS 2.2.0 a4).   I don't have Prusa Slicer installed on a PC, but I'm happy to do so if that helps.
[image]
[image]

coolhoun, 2020-02-20:
Also, one detail I neglected to mention, is that I do change the "Purging volumes...".  Not sure if that has any relation.

bubnikv, 2020-02-21:
Would you please try to find as short as possible steps to reproduce the
issue, starting with a freshly opened PrusaSlicer?

coolhoun, 2020-02-21:
Please disregard my comment "re: Purging Volumes", as I think this has just added confusion.  I did **_not_** edit this settings when reproducing the steps.  The steps I listed above are _**exactly _** what I did to reproduce the issue.  Later today, I will install Prusa Slicer fresh on another MacBook and run through these steps and report the results.**_**_

bubnikv, 2020-03-20:
It looks there were two issues mixed up here. The initial issue
https://github.com/prusa3d/PrusaSlicer/issues/2540#issuecomment-517460417
has been hopefully fixed with a9fc39491e0b37f9a68ac99c372f09121e4c8afd

The issues in PrusaSlicer 2.2.0 alphas, betas and release candidates were described in #3842 and fixed in PrusaSlicer 2.2.0-rc5 with 98fb8b626fa09c6196f3afda1fc4ee827e5e08dc. The issues in PrusaSlicer 2.2.0 were due to refactoring, so they were new in PrusaSlicer 2.2.0 and they were not there in PrusaSlicer 2.0 and 2.1. 

I am closing this issue. Please feel free to reopen if this issue shows up in PrusaSlicer 2.2.0 final release.

closed 2020-03-20 by bubnikv