prusa3d/PrusaSlicer #2541: First Layer Height bug on reloading .stl

richiz86, 2019-06-21:
### Version
PrusaSlicer 2.0.0+w64

### Operating system type + version
Win10 pro

### 3D printer brand / version + firmware version (if known)
MK3S newest
Profile: 0.2 DETAIL (no changes)

### Behavior
First Layer Height bug on reloading .stl

I created a test to reproduce the bug:

I created a test cube in Fusion 360 with a side length of 25 mm an imported the STL in PrusaSlicer 2.0. The testcube was created with a square sketch on the x-y plane and extruded 25mm in the z direction. At this point everything worked. Now I go back into Fusion 360 and change the extrusion height in Fusion 360 to 24 mm, so cube becomes 1 mm shorter in height. After exporting the file into an STL I reimport the STL with rightclick - reload from disk. Now the first layer height in Prusa Slicer is Set to 0.8 mm, and can not be lowered. Only removing an reloading the part helped.

Something similar happens when you start with a smaler model and make it taller. The Difference is that several layers at the top of the Part are getting removed

When I make the height difference bigger than 1mm, PrusaSlicer crashes.

Forum disscussion:
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/prusa-slicer-first-layer-height-bug/#post-148951

stuartpb, 2019-06-23:
Yeah, I came here to report the same issue: model height isn't recalculated when hitting "Reload from Disk". The model's Z position on the build plate should be re-evaluated when reloading from disk, with it being put back on the build plate if the height has changed. (I had a model that I reduced in height by 2mm, and I didn't understand why it was mis-printing until I noticed that the first layer was starting at 1mm off the plate.)

haarp, 2019-07-21:
Can confirm. If you don't notice, the print will even start in the air, with potentially fatal extruder blobs as a consequence.

Public Service Announcement: Don't be like me, supervise the first two or so layers of your prints ;)

rtyr, 2021-02-02:
Should be fixed. Closing.

closed 2021-02-02 by rtyr