prusa3d/PrusaSlicer #2542: Slicing text does not work right.

tiendaaep, 2019-06-22:
### Version
_Version of PrusaSlicer used goes here_
I found the problem in all versions. Now in 2.0.0
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
MK3S
### Behavior
When i slice text that have thin lines, some lines does not appear after slicing. If the letters are big it works ok but i have letters 3 mm high and the text is incomplete. It works ok if i slice with CURA. I Enabled Thin lines in PrusaSlic3r and does not work.
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[Parametros CURA.docx](https://github.com/prusa3d/PrusaSlicer/files/3316551/Parametros.CURA.docx)

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs

When i try to add an .RMF file it says that is not supported.

neophyl, 2019-06-22:
rename your file from .3mf to .zip that way the software will accept it as a valid file type

tiendaaep, 2019-06-22:
Hi, i send the rmf enclosed in this email.
Is it ok or i need to send by GitHub?
Thank you.
Best regards.
Isidoro.

bubnikv, 2019-06-24:
You may as well try @supermerill 's fork. He did a lot of work on the gap fill / thin lines algorithm, which we plan to review and integrate.
https://github.com/supermerill/Slic3r/releases

tiendaaep, 2019-06-24:
I have downloaded the version that do it and unfortunately it does not
work.  I have windows 10 and when i slice any model it looks like that only
slice the firt layer and allways says that my print will took 29 seconds.

closed 2026-09-29 by prusaslicer-bot[bot]