prusa3d/PrusaSlicer #2543: Prusa 2.0.0 Z axis Motors Turning Slowly

Wallys3d, 2019-06-22:
### Version
_Version of 2.0 used goes here_

_Use `About->About Prusa 2.0.0+win64 for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_windows 8.1 64 
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)Folgertech 2020 i3 Marlin
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_Z axis Steppers barely turning when Homing loud noise, When using Repetier Host this does not Happen
* _Steps needed to reproduce the problem_ Slice the Model, Save to sd card, tell it to Print 
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2019-06-24:
PrusaSlicer produces machine limits G-code at the start for the Marlin
firmware. Maybe your Marlin firmware interprets these numbers differently?
Try to switch to Reprap / Sprinter and this G-code section will not be
generated.

Wallys3d, 2019-06-25:
> rusaSlicer produces machine limits G-code at the start for the Marlin firmware. Maybe your Marlin firmware interprets these numbers differently? Try to switch to Reprap / Sprinter and this G-code [...]

Thank-you I had no clue I really appreciate your information, i`ll try to figure out how to do  your suggestion and i`ll let you know.

Wallys3d, 2019-06-25:
Hello: bubnikikv

All i had to do was comment out
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
;M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec

everything is working fine, Thanks to you i can start learning this new Prusa 2.0.0 slicer
Thank-you for taking the time to offer your suggestion this newbie really appreciates it

closed 2019-06-25 by Wallys3d