prusa3d/PrusaSlicer #2544: No pause icon in layers preview view if printer is configured as multimaterial

Goury, 2019-06-22:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2019-06-24:
The plus icon is not shown because the behavior is more complex than with single extruder printers and it was not yet implemented. And given that the toolchange can be enforced by a modifier mesh, this is not considered especially high priority task. See also https://github.com/prusa3d/PrusaSlicer/issues/1725.

Inserting a custom gcode instead of M600 was already suggested too (https://github.com/prusa3d/PrusaSlicer/issues/2351, https://github.com/prusa3d/PrusaSlicer/issues/1899, https://github.com/prusa3d/PrusaSlicer/issues/1656). This is something that we will hopefully implement, but not before we deal with more pressing features and bugs.

DieKatzchen, 2019-06-27:
Can it be explained to me, simply, how "the toolchange can be enforced by a modifier mesh" because even though this is your advice every time it comes up, I can find no information on how to actually do it.

DieKatzchen, 2019-06-27:
Actually, no. Enforcing a toolchange doesn't help. There's already a toolchange because it's multimaterial. I want to add an m600 on a SPECIFIC toolchange, not every toolchange. If it pauses on every toolchange I might as well not have multiple extruders. If I want to print a three color object that has been carefully designed to only have two colors at a time I should only have to manually change filament ONCE.

lukasmatena, 2019-06-27:
@DieKatzchen If that's what you need, then I can only repeat what I wrote already - it is not yet implemented. Post-processing the gcode is probably the only option right now.

lukasmatena, 2020-03-30:
@DieKatzchen I believe that what you wanted is now possible in PrusaSlicer2.2.0.
Closing the issue. If you disagree, we can reopen it.

closed 2020-03-30 by lukasmatena