prusa3d/PrusaSlicer #2545: temperature from first extruder always used

gnydick, 2019-06-22:
### Version
2.0.0+win64

### Operating system type + version
Windows10

### Bug
I have a dual extruder printer. When I use only the right tool (#2) the temperature from the filament chosen for the left tool (#1) is used to set the nozzle temp. The bed temp is correct.

ext-1 - prusament 215
ext-2 - ABS 255

Here is the excerpt
```

M107
M561 ;Clear the mesh level info (will load below)
M375 ; load stored height map
G29 S1; enable bed mesh

T1
M109 S215
M190 S110

;Intro Line Near Right
G1 X0 Y300 Z20 F2000
G1 X0 Y280 Z.3 E2 F3000

```

bubnikv, 2019-06-24:
If you have a dual extruder, you should disable the "single extruder multi
material" feature. In that case slicer will set temps for both the
extruders at the start of the print.

lukasmatena, 2019-08-01:
@gnydick I'm sorry, I cannot reproduce the problem. Could you please enclose 3MF that would allow me to? Thanks.

lukasmatena, 2019-10-09:
This is similar to #2719.

I didn't manage to reproduce either of the issues and there is no response from their authors after a 3MF was requested. There also were two similar bugs, both fixed for 2.1.0 (commits 5393979 and 170bd8b), maybe this was flushed along. Could have also been a user mistake.

Closing.

closed 2019-10-09 by lukasmatena