prusa3d/PrusaSlicer #2546: Changing infill pattern setting hangs slic3r for some time

Goury, 2019-06-22:
### Version
2.0.0+linux64

### Behavior
* Load some STL file
* Slice it
* Don't like how infill looks
* Go to print settings → infill
* Change "Fill pattern" to something else
* It hangs for a minute
* Now I finally can reslice this thing

### Expected behavior
* No hangups, just need to reslice.

[TPHC.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3317140/TPHC.3mf.zip)

Goury, 2019-06-22:
UPD: may had been caused by slic3r automatically preparing gcode for export, even though no one asked it to prepare it.
How do I tell it to stop doing this?

BerndJM, 2019-06-22:
Configuration -> Preferences -> Background Processing

Goury, 2019-06-22:
Seems like this new slic3r is as buggy as cura
[image]

bubnikv, 2019-06-24:
The background processing needs to check regularly for cancellation at
cancellation points. It may be, that we did not cover some of the
algorithms with enough of these cancellation points. It would help us, if
you could identify which algorithm it is hanging on for a single minute. My
guess is it is the "avoid crossing perimeters" feature.

You would help us if you could run slicer with the --loglevel=5 parameter
(you may experiment with the level number) and find out, at which point the
application hangs.

bubnikv, 2019-06-24:
You have the "avoid crossing perimeters" feature enabled. You may try to
disable it.

bubnikv, 2019-08-25:
Aborting issue of the "avoid crossing perimeters" feature has been fixed with 3addae3, and the feature is now also significantly faster. Closing.

closed 2019-08-25 by bubnikv

Goury, 2019-08-27:
@bubnikv why did you close this one?
The problem seems to be not in the "avoid crossing perimeters" setting but rather in Slicer not cancelling slicing that is in progress and instead just hanging.

bubnikv, 2019-08-27:
Did you test the beta2 and did you read the beta2 change log? The
cancellation issue should be solved.

Goury, 2019-08-27:
Not yet, but I will today