prusa3d/PrusaSlicer #2547: Retraction Distance Induced Blobbing

MeltBurnMake, 2019-06-22:
### Version
PrusaSlicer-2.0.0

### Operating system type + version
Windows 7

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3, firmware 3.7.0

### Behavior
When printing with increased retraction distances, the printer will pause at certain points, the screen will read "heating," and a blob of plastic will push out (possibly ooze, but it looks like a lot more). The print will then continue, until it repeats with another pause and blob.

This was noticed when I tried using settings from Slic3r PE (last version before 2.0 became available) to print wood PLA. For this material, I have a particular set of settings for print, filament, and printer, to eliminate stringing. Upon porting them into the Prusa Slicer 2.0, using the wood PLA settings resulted in horrible pause and blobs.

To see what the cause might be, I left the print settings and filament settings the same (ones designed for wood filament) but rolled back the printer settings to one I use for most PLA. My intent was to rotate through settings until the problem disappeared, but it went away after this first step.

The largest difference between settings is the retraction distance, which for me is 7mm. I set up a new printer setting file, and started printing retraction tests at greater retraction distances until the issue appeared again. At 0.4mm, nothing. 1mm, nothing. 3mm, I had 1 blob half way up the print. At 7mm (aimed at recreating the original settings) it was almost entirely blobs.

It appears that at increased retraction distances, something else in the software is triggering something that tells the printer to pause and heat.

This was not an issue on the old slicer.

images: 0.4mm retraction, 3mm retraction, and 7mm retraction

[image]
[image]
[image]

Here's a copy of the gcode for the retraction test at 7mm retraction distance (60mm/s retraction speed)
[Retraction Test.txt](https://github.com/prusa3d/PrusaSlicer/files/3317173/Retraction.Test.txt)

Thank you for your help, let me know if anything doesn't make sense.
~MeltBurnMake

bubnikv, 2019-06-24:
I have asked the testing team to verify this issue.

DRracer, 2019-06-24:
Dear MeltBurnMake, the testing team was able to reproduce your issue with 7mm retract and the "heating-stop" using your GCODEs. It seems to be somehow related to the optical filament sensor in MK3, we will investigate it further in the future in the firmware development team. We haven't been able to reproduce these probems on a MK3S (different filament sensor). So far, it looks unrelated to PrusaSlicer itself.

bubnikv, 2019-08-01:
Firmware issue, closing.

closed 2019-08-01 by bubnikv