prusa3d/PrusaSlicer #2548: Error when changing nozzle size with multiple extruders (MMU)

iamgnat, 2019-06-22:
### Version
2.0.0+linux64

### Operating system type + version
Ubuntu 18.04

### 3D printer brand / version + firmware version (if known)
Prusa MK3S+MMU2S 3.7.0/1.0.5

### Behavior

When changing the nozzle size for an extruder an error pops up stating `All extruders must have the same diameter for single extruder multimaterial printer.`. You then get this same error when you change the nozzle size for all but the last extruder (because then they all match).

Since it knows that it is using a single extruder I would expect one of the following behaviors:

- That the nozzle size is moved to a central location so it is a single change rather than for each extruder.
- If it has to stay per extruder:
  - That it changes all when one is changed
  - It prompts the user with an offer to apply the change to all extruders.
- After the first warning it only warns again if the size doesn't match the size for another nozzle.
  - This would obviously need a check at slice time to make sure all extruders match.

_Is this a new feature request?_ Yes

bubnikv, 2019-06-24:
A reasonable solution would be to ask "Do you want to change the diameter for all extruders" yes / no.
@YuSanka would you please take care of it? Thanks.

FidelCapo, 2019-06-26:
Thank you for report. This issue should be fixed in next release.
Closing.

closed 2019-06-26 by FidelCapo