prusa3d/PrusaSlicer #2549: Load gcode feature

gaellafond, 2019-06-23:
### Version
PrusaSlicer 2.0.0+linux64

### Operating system type + version
Debian GNU/Linux 9

### 3D printer brand / version + firmware version (if known)
Genuine Prusa i3 MK3, firmware 3.7.1

### Behavior
New feature request

In PrusaControl, you can load gcode files and preview them:
File > Import gcode file
It's far from perfect, but it's quite useful to do a quick assessment of the gcode.
This feature is missing from PrusaSlicer

It's very useful to preview downloaded gcode files before sending them to your printer, to be sure they are what you expect. Is the print too big? Printing in mid-air? Printing a different parts then the one downloaded? Maybe the user was trolling and he uploaded the gcode of an offensive 3D model instead of the expected one. Since PrusaPrinters repository allow users to upload gcode file, that feature is becoming an increasingly important security feature.

#### Project File (.3MF) where problem occurs
N/A

ManuGithubSteam, 2019-06-24:
i second this :-)

rtyr, 2021-02-02:
Implemented in 2.3.0. Closing.

closed 2021-02-02 by rtyr