prusa3d/PrusaSlicer #2550: non-Prusa printer causes "Slice now" to hang during estimating material/time

jp-powers, 2019-06-23:
### Version
2.0.0+linux64

### Operating system type + version
Linux Mint 19.1 64bit (Cinnamon)

### 3D printer brand / version + firmware version (if known)
FLSun QQ-S

### Behavior
Printer is added into PrusaSlicer, print settings are established, load models, hit Slice now, and PrusaSlicer does the slicing but eventually hangs. When viewing the preview it appears like the object is sliced properly but the loading bar is stuck at nearly complete. The only thing that doesn't appear finished is the material and time estimates.

[image]

#### Project File (.3MF) where problem occurs
[PrusaSlicerHangsWithNonPrusaPrinter3MF.zip](https://github.com/prusa3d/PrusaSlicer/files/3318688/PrusaSlicerHangsWithNonPrusaPrinter3MF.zip)

neophyl, 2019-06-24:
I tried to slice it and had the same issue as you.  I then noticed you had avoid crossing perimeters turned on in print settings.  This is a known factor that can make Slicer take forever to actually finish slicing.  So I turned it off and tried slicing it again.

Now it gives me the following error :- 
[image]

Id attach the temp file but its over 30mb.  it does have this at the top though

; generated by PrusaSlicer 2.0.0+win64 on 2019-06-24 at 08:32:38

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

M201 X1000 Y1000 Z1000 E1000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z200 E200 ; sets maximum feedrates, mm/sec
M204 P1000 R1000 T1000 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X20.00 Y20.00 Z0.40 E5.00 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M190 S60 ; set bed temperature and wait for it to be reached
M104 S215 ; set temperature

!!!!! Failed to process the custom G-code template start_gcode
Parsing error at line 1: Variable does not exist
;M190 S[print_bed_temperature] ;Uncomment to add your own bed temperature line
        ^
!!!!! End of an error report for the custom G-code template start_gcode

neophyl, 2019-06-24:
btw I turned on the avoid crossing perimeters option again with the modified start/end gcode and left it slicing.  It took about 20 minutes to complete but it did complete

bubnikv, 2019-06-25:
print_bed_temperature is a vector variable. Best is to use  the new
placeholder syntax (in curly braces)
https://github.com/prusa3d/PrusaSlicer/wiki

M190 S[print_bed_temperature]
should be written as
M190 S{print_bed_temperature[initial_tool]}
or for example
M190 S[print_bed_temperature_0]

jp-powers, 2019-06-27:
Thanks for the assist, it looks like combining the new placeholder syntax with simply disabling avoid crossing perimeters was enough to get it to actually slice. Now I just gotta find out how well it works on the actual printer.

The placeholder syntax is a carry over from some FLSun QQ-S user group's agreed upon start/end g-code for Cura. I'm just cutting out all of those variables, specifically as it was commented out anyways and going unused.

As for the avoiding crossing perimeters, this is for a bowden based Delta printer, but I'm guessing I saw the "problem" as I'm running PrusaSlicer on a Dell XPS 13 9370. I let it sit for ~30 minutes once on a small/very low poly count object, saw no progress, assumed it hung. Guessing if I tried it on my gaming desktop I'd have seen it finish. I'll disable it for now to confirm the basic profile is working, and re-enable it if it proves necessary.

Anyways, it completed, so I'm happy to close this as it's not a real bug. Thanks so much for the assistance!

closed 2019-06-27 by jp-powers