prusa3d/PrusaSlicer #2552: Feature req: allow changes to fan speed via modifier mesh

icanhazpython, 2019-06-24:
I would love the option for changing fan speed via a modifier mesh. I recently printed a functional part in petg that is subjected to high stress. I turned off the fan to maximize layer adhesion, however no fan on petg makes for poor overhangs. As my part had a hole going horizontally through it, I ended up writing some layer change gcode that would turn the fan on for only the layers near the top of the hole. This helped a great deal. I think being able to vary fan speed in the area of a modifier mesh could allow me to obtain maximum layer adhesion with petg, while still getting decent overhangs where I need them.

bubnikv, 2019-06-24:
We are finalizing an effort of allowing to define layer height per layer
span as in the Slic3r PE or the upstream Slic3r. In addition, we are
allowing all the parameters you can assign to a modifier mesh to be
assigned to a layer span. We are considering to add the g-code export
parameters as the temperatures or fan speed to the layer span as well.

It would be much more complex to add these parameters to general modifier
meshes, but it may be a reasonable effort to allow it for the layer spans.

icanhazpython, 2019-09-01:
> We are finalizing an effort of allowing to define layer height per layer span as in the Slic3r PE or the upstream Slic3r. In addition, we are allowing all the parameters you can assign to a modifier [...]

Thanks, I saw the latest release notes included the layer span feature. I have not tested yet, but if it supports setting fan speed for a given layer span, that would be great.

bubnikv, 2019-09-02:
>  if it supports setting fan speed for a given layer span, that would be
great.

No, it does not yet.

rtyr, 2021-02-02:
Same as https://github.com/prusa3d/PrusaSlicer/issues/3439. I will close this one.

closed 2021-02-02 by rtyr