prusa3d/PrusaSlicer #2554: [FeatureRequest] Display real meters of filament used additionally to the meters pressed out of the nozzle.

ManuGithubSteam, 2019-06-24:
Hi
I'm kinda a technical person an i can easy guess how much meters are left of the spool.
However if say roughly one meter is left.

Then i print something and it says 4 meters used in the slicer. HOWEVER these 4 meters are pressed 4 meters trough a 0.4 nozzle so in reality i have to divide that with 4.3 and i get my meter. So i probably can print that part.

I propose this output format:

Used Filament 1.75mm: 1 meter
Used Filament 0.4 mm nozzle: 4 meters

So its easily guessable how much filament you need on your spool ans also you get the the info how much this becomes when it is pressed tough the nozzle.

Thanks

lukasmatena, 2019-06-24:
I'm sorry, what? PrusaSlicer of course shows actual length of the filament, not the length of what goes through the nozzle (change the nozzle diameter to see the estimate does not change if you cannot do the math yourself).

> I'm kinda a technical person

If you were, you would not divide by 1.75/0.4=4.3, but by a square of that. That's how the two lengths would actually differ.

ManuGithubSteam, 2019-06-24:
> I'm sorry, what? PrusaSlicer of course shows actual length of the filament, not the length of what goes through the nozzle (change the nozzle diameter to see the estimate does not change if you cannot do the math yourself).

Thanks for clarifying that :-) You are clearly right.

> 
> > I'm kinda a technical person
> 
> If you were, you would not divide by 1.75/0.4=4.3, but by a square of that. That's how the two lengths would actually differ.
Well i normally work with wood and just recently with 3d printers.

closed 2019-06-24 by ManuGithubSteam