prusa3d/PrusaSlicer #2555: Black control groups when rebuilding on OSX Mojave

wavexx, 2019-06-24:
### Version
Rebuilt from master @ b8f9d2e2d8b312f3dbf55aed05291a70d4e92d30

### Operating system type + version
OSX Mojave 10.14.5

### Behavior
I attempted a rebuild from scratch on OSX following the instructions (https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Mac%20OS.md)

"deps" built fine after two tweaks:

- Installed gettext from homebrew (msgfmt was required during the build of wxWidgets).
- Skipped the cli binary of qhull (failed to link for some reason - but since we don't need it I just touched the executable and went on)

After that PrusaSlicer itself built fine and without problems.
But when running, the various setting tabs seen to have a black background in the option groups.

The first group in a tab is always fine, with the groups following always having a black background:

[image]

[image]

Any idea why this is the case?

Everything seems functional. If you change a setting the labels become orange, so the text is there but just invisible due to the background. 

The system is running in the default "light"mode.

Ping @Quintox303

vojtechkral, 2019-06-24:
Could you post your `cmake` outputs from both the deps build as well as the slicer build? (Just the `cmake` output should be fine, no need to include full build logs including `make`) I'm suspecting this might be due to bad Mac OS target versions.

Quintox303, 2019-06-26:
@vojtechkral you are right that I was not targeting the correct OS version. I managed to set this to 10.9 as the instructions state, but then I run into this error when I run `make` for the dependencies.

[image]

Which seems to be the same issue here that you commented on https://github.com/glennrp/libpng/issues/187. What do I need to change to get past this?

vojtechkral, 2019-06-27:
@Quintox303 I have the same advice for you as for OP - please post `cmake` output :-) `cmake` outputs bunch of useful info such as compiler used, target version, SDK path etc. etc., from which I'll be hopefully able to glean some insight...

wavexx, 2019-06-30:
@vojtechkral I was reporting on behalf of @Quintox303 for this issue.

Quintox303, 2019-07-01:
@vojtechkral Here are my cmake outputs. Hopefully this can help diagnose the issue. I managed to get past that error that my last comment was about, but the black control groups are still present. Here are the outputs:

Cmake of dependencies

<code>
-- The C compiler identification is AppleClang 10.0.1.10010046
-- The CXX compiler identification is AppleClang 10.0.1.10010046
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- PrusaSlicer deps DESTDIR: /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir
-- PrusaSlicer deps debug build: ON
OS X SDK Path: /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk
OS X Deployment Target: 10.9
-- Found Git: /usr/bin/git (found version "2.20.1 (Apple Git-117)") 
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/matthew/Downloads/PrusaSlicer-master2/deps/build
</code>

Cmake of PrusaSlicer

<code>
-- The C compiler identification is AppleClang 10.0.1.10010046
-- The CXX compiler identification is AppleClang 10.0.1.10010046
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to Release
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: OFF
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_SYNTAXONLY: OFF
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: /Users/matthew/Downloads/PrusaSlicer-master2/buikd/../deps/build/destdir/usr/local (from cache or command line)
OS X SDK Path: /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk
OS X Deployment Target: 10.9
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Boost version: 1.66.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   log
--   locale
--   regex
--   chrono
--   date_time
--   atomic
--   log_setup
-- Boost::boost exists
-- Found TBB: /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/include (found version "2017.0")  
--   TBB_FOUND               = TRUE
--   TBB_INCLUDE_DIRS        = /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/include
--   TBB_DEFINITIONS         = 
--   TBB_LIBRARIES           = /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libtbb_static.a
--   TBB_DEFINITIONS_DEBUG   = -DTBB_USE_DEBUG=1
--   TBB_LIBRARIES_DEBUG     = 
--   TBB_DEFINITIONS_RELEASE = 
--   TBB_LIBRARIES_RELEASE   = /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libtbb_static.a
-- Found CURL: /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libcurl.a (found version "7.58.0") 
-- Miniz NOT found in system, using bundled version...
-- qhull Version: 7.2.0 (static linking)
-- IGL found, using system version...
-- Parallelization with Intel TBB
-- Found NLopt in '/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib'.
-- Using NLopt include directory '/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/include'.
-- Using NLopt library '/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libnlopt.a'.
-- Found GTest: /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libgtest.a (Required is at least version "1.7") 
-- Found wxWidgets: -L/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib;;;-framework IOKit;-framework Carbon;-framework Cocoa;-framework AudioToolbox;-framework System;-framework OpenGL;/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_gl-3.1.a;/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_html-3.1.a;/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_adv-3.1.a;/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_core-3.1.a;/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libwx_baseu-3.1.a;-framework OpenGL;-framework AGL;-framework WebKit;-lwxtiff-3.1;-lwxjpeg-3.1;-lwxpng-3.1;-lwxregexu-3.1;-lwxscintilla-3.1;-lwxexpat-3.1;-lz;-framework Security;-lpthread;-liconv (found suitable version "3.1.1", minimum required is "3.1") 
-- wx-config path: /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/bin/wx-config
-- Found ZLIB: /usr/lib/libz.dylib (found version "1.2.11") 
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/matthew/Downloads/PrusaSlicer-master2/buikd
</code>

vojtechkral, 2019-07-01:
@Quintox303 I took the liberty and edited your comment, fixing formatting for clarity. (No content changed.)

In any case, here's the problem:

```
OS X SDK Path: /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk
OS X Deployment Target: 10.9
```

The deployment target doesn't match the SDK used. I'm not sure how this happened, but you probably want to build for newer Mac OS 10.14. To do this, set `CMAKE_OSX_DEPLOYMENT_TARGET` to `10.14` (on the command line, use the flag `-DCMAKE_OSX_DEPLOYMENT_TARGET=10.14` ).

Alternatively if you wanted to build for 10.9, you'd need to download the appropriate SDK.

Hope this helps.

lekv, 2019-07-13:
I'm observing the same issue. Here's the output of cmake for deps:

```
$ cmake ..
-- The C compiler identification is AppleClang 10.0.1.10010046
-- The CXX compiler identification is AppleClang 10.0.1.10010046
-- Check for working C compiler: /usr/local/opt/ccache/libexec/cc
-- Check for working C compiler: /usr/local/opt/ccache/libexec/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/local/opt/ccache/libexec/c++
-- Check for working CXX compiler: /usr/local/opt/ccache/libexec/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- PrusaSlicer deps DESTDIR: /Users/lv/checkout/PrusaSlicer/deps/build/destdir
-- PrusaSlicer deps debug build: ON
OS X SDK Path: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk
OS X Deployment Target (inferred from default): 10.14
!!! Including unix deps
-- Found Git: /usr/local/bin/git (found version "2.18.0")
!!!! In the unix file
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/lv/checkout/PrusaSlicer/deps/build
```

Here's the output of cmake for prusaslicer itself:
```
$ cmake .. -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local" -DCMAKE_OSX_DEPLOYMENT_TARGET=10.14 2>&1 | tee ../cmake.log
-- The C compiler identification is AppleClang 10.0.1.10010046
-- The CXX compiler identification is AppleClang 10.0.1.10010046
-- Check for working C compiler: /usr/local/opt/ccache/libexec/cc
-- Check for working C compiler: /usr/local/opt/ccache/libexec/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/local/opt/ccache/libexec/c++
-- Check for working CXX compiler: /usr/local/opt/ccache/libexec/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to Release
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: OFF
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_SYNTAXONLY: OFF
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: /Users/lv/checkout/PrusaSlicer/build/../deps/build/destdir/usr/local (from cache or command line)
OS X SDK Path: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk
OS X Deployment Target: 10.14
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Boost version: 1.66.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   log
--   locale
--   regex
--   chrono
--   date_time
--   atomic
--   log_setup
-- Boost::boost exists
-- Found TBB: /Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/include (found version "2017.0")  
--   TBB_FOUND               = TRUE
--   TBB_INCLUDE_DIRS        = /Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/include
--   TBB_DEFINITIONS         = 
--   TBB_LIBRARIES           = /Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/lib/libtbb_static.a
--   TBB_DEFINITIONS_DEBUG   = -DTBB_USE_DEBUG=1
--   TBB_LIBRARIES_DEBUG     = 
--   TBB_DEFINITIONS_RELEASE = 
--   TBB_LIBRARIES_RELEASE   = /Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/lib/libtbb_static.a
-- Found CURL: /Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/lib/libcurl.a (found version "7.58.0") 
-- Miniz NOT found in system, using bundled version...
-- Using qhull from system.
-- IGL found, using system version...
-- Parallelization with Intel TBB
-- Found NLopt in '/Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/lib'.
-- Using NLopt include directory '/Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/include'.
-- Using NLopt library '/Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/lib/libnlopt.a'.
-- Found GTest: /Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/lib/libgtest.a (Required is at least version "1.7") 
-- Found wxWidgets: -L/Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/lib;;;-framework IOKit;-framework Carbon;-framework Cocoa;-framework AudioToolbox;-framework System;-framework OpenGL;/Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_gl-3.1.a;/Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_html-3.1.a;/Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_adv-3.1.a;/Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_core-3.1.a;/Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/lib/libwx_baseu-3.1.a;-framework OpenGL;-framework AGL;-framework WebKit;-lwxtiff-3.1;-lwxjpeg-3.1;-lwxpng-3.1;-lwxregexu-3.1;-lwxscintilla-3.1;-lwxexpat-3.1;-lz;-framework Security;-lpthread;-liconv (found suitable version "3.1.1", minimum required is "3.1") 
-- wx-config path: /Users/lv/checkout/PrusaSlicer/deps/build/destdir/usr/local/bin/wx-config
-- Found ZLIB: /usr/lib/libz.dylib (found version "1.2.11") 
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/lv/checkout/PrusaSlicer/build
```

I'm on MacOS 10.14.5.

Quintox303, 2019-07-15:
I rebuilt again with the deployment target corrected and I have the same results. I am also on MacOS 10.14.5. Here are my cmake outputs

deps:
```
-- The C compiler identification is AppleClang 10.0.1.10010046
-- The CXX compiler identification is AppleClang 10.0.1.10010046
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- PrusaSlicer deps DESTDIR: /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir
-- PrusaSlicer deps debug build: ON
OS X SDK Path: /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk
OS X Deployment Target (inferred from default): 10.14
-- Found Git: /usr/bin/git (found version "2.20.1 (Apple Git-117)") 
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/matthew/Downloads/PrusaSlicer-master2/deps/build
```

prusaslicer:
```
-- The C compiler identification is AppleClang 10.0.1.10010046
-- The CXX compiler identification is AppleClang 10.0.1.10010046
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to Release
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: OFF
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_SYNTAXONLY: OFF
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: /Users/matthew/Downloads/PrusaSlicer-master2/build/../deps/build/destdir/usr/local (from cache or command line)
OS X SDK Path: /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk
OS X Deployment Target: 10.14
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Boost version: 1.66.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   log
--   locale
--   regex
--   chrono
--   date_time
--   atomic
--   log_setup
-- Boost::boost exists
-- Found TBB: /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/include (found version "2017.0")  
--   TBB_FOUND               = TRUE
--   TBB_INCLUDE_DIRS        = /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/include
--   TBB_DEFINITIONS         = 
--   TBB_LIBRARIES           = /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libtbb_static.a
--   TBB_DEFINITIONS_DEBUG   = -DTBB_USE_DEBUG=1
--   TBB_LIBRARIES_DEBUG     = 
--   TBB_DEFINITIONS_RELEASE = 
--   TBB_LIBRARIES_RELEASE   = /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libtbb_static.a
-- Found CURL: /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libcurl.a (found version "7.58.0") 
-- Miniz NOT found in system, using bundled version...
-- qhull Version: 7.2.0 (static linking)
-- IGL found, using system version...
-- Parallelization with Intel TBB
-- Found NLopt in '/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib'.
-- Using NLopt include directory '/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/include'.
-- Using NLopt library '/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libnlopt.a'.
-- Found GTest: /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libgtest.a (Required is at least version "1.7") 
-- Found wxWidgets: -L/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib;;;-framework IOKit;-framework Carbon;-framework Cocoa;-framework AudioToolbox;-framework System;-framework OpenGL;/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_gl-3.1.a;/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_html-3.1.a;/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_adv-3.1.a;/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libwx_osx_cocoau_core-3.1.a;/Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/lib/libwx_baseu-3.1.a;-framework OpenGL;-framework AGL;-framework WebKit;-lwxtiff-3.1;-lwxjpeg-3.1;-lwxpng-3.1;-lwxregexu-3.1;-lwxscintilla-3.1;-lwxexpat-3.1;-lz;-framework Security;-lpthread;-liconv (found suitable version "3.1.1", minimum required is "3.1") 
-- wx-config path: /Users/matthew/Downloads/PrusaSlicer-master2/deps/build/destdir/usr/local/bin/wx-config
-- Found ZLIB: /usr/lib/libz.dylib (found version "1.2.11") 
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/matthew/Downloads/PrusaSlicer-master2/build
```

vojtechkral, 2019-07-15:
Ok, thanks, I'll have a look...

DrLex0, 2019-07-17:
I have the same issue. My workaround is to set up the print in the regular release, then save the project and open it in my custom build. A fix would be highly appreciated!

vojtechkral, 2019-07-22:
Allright, so I managed to reproduce the issue. It seems to happen both with wxWidgets 3.1.1-patched and 3.1.2-vanilla. At the moment I have no idea what causes this. It doesn't seem to be a dark mode issue. Chances are this is a problem in wxWidgets, but I can say at the moment.

DrLex0, 2019-12-23:
This seems fixed now, I no longer have black areas when building in 10.14.6 against the default 10.14 SDK (building against 10.9 is broken by the way, some error about exceptions only being supported from 10.12 on).

rtyr, 2021-02-02:
Should be fixed. Closing.

closed 2021-02-02 by rtyr