prusa3d/PrusaSlicer #2556: Add Unidirectional INFILL

GWdd, 2019-06-24:
### Version
2.0.0 Win64

### Operating system type + version
Windows 7 Pro x64

### 3D printer brand / version + firmware version (if known)
Prusa Mk3

### Behavior
Solid Infill - e.g., top and bottom layers, often has holes and other features that interrupt a continuous infill. Plicer will fill up to these points, continue around them, and then reverse direction to fill areas that were missed. 

This leads to unusual looking lines and artifacts where the two infill passes meet. 

I would like a setting to force Plicer to only perform infill in a single direction to help eliminate these print artifacts.

This part and attached gcode shows how Plicer prints the bottom layer two directions.  While not the worst example, it shows a condition where I'd like the ability to force unidirectional infill. Basically, what is happening is shown in this sketch:
[image]

- NOTE: I am asking for infill to start at front - stroke left and right towards back. And if needed, move to front side of next area to be filled, stroke left and right toward back. 

Is this a new feature request? Yes and no; I almost consider the artifacts caused by the default infill method a defect.

#### Project File (.3MF) where problem occurs
--- Attached stl, gcode, and 3mf
[InfillExample.zip](https://github.com/prusa3d/PrusaSlicer/files/3322506/InfillExample.zip)

benkayuk, 2019-07-04:
Are you sure you've uploaded the correct example file? Doesn't exhibit this issue as it's a flat base with no holes in?

GWdd, 2019-07-04:
Yep - wrong half of the project ... here's a better example.
[image]
[InfillPattern.zip](https://github.com/prusa3d/PrusaSlicer/files/3360286/InfillPattern.zip)

GWdd, 2019-07-06:
Why was this closed? It certainly isn't addressed.

bobstro, 2019-07-08:
It says you closed it and re-opened it. Quick test would be to rotate by a few degrees.

GWdd, 2019-07-08:
I've never closed anything on github, especially this topic ... lol.  As a matter of fact, I don't have a clue how to close a topic.  Unless it's this huge button at the bottom that says [CLOSE AND COMMENT] and I clicked that thinking all it does is close the post window.

wlcina, 2019-09-30:
Hello how far are we with solution of this? I understand there may not be an ideal case without artifacts, but I believe minimize of artifacts is possible - by filling largest area fill at once, then smaller and smaller.  See in the picture (follow the gradient from light to dark green):
[image]

rtyr, 2020-11-18:
Implemented in 2.3.0-alpha (monotonic infill). Closing.

closed 2020-11-18 by rtyr